greatscottgadgets/hackrf #473: Loss of quality recording dvb-t

xugmu, 2018-04-11:
### Steps to reproduce
1.Record  and  replay dvb-t  channel  8mhz
2.
3.

### Expected behaviour
The same quality as the original signal

### Actual behaviour
7 db less in modulation error rate

### Version information
Ubuntu 16.04 LTS

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82566ad5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00474f57

If you are reporting a problem that involves third party software 

Any program

### Output
```
I

Best regards
```

xugmu, 2018-04-15:
I'm going to answer myself

The hackrf seems to be comfortable working at 10mhz. A loss of 6 or 7 db of modulation error rate in an A-D process D-A may be considered normal. We are talking about an 8K ODFM channel

Luckily the cancellation of the DC offset in gqrx seems to work well

Best  regards

dominicgs, 2018-04-23:
Using the bandwidth filter and sample rate that is closest (but greater than) your signal bandwidth is likely to give you the best results.

As you've found here, using the 10MHz filter and rate for an 8 MHz signal is best.

closed 2018-04-23 by dominicgs