greatscottgadgets/hackrf #474: Calibratin

pachoduran, 2018-04-12:
Hi, sorry for me english.

Hello, I need your help to solve a problem. I'm doing a simple spectrum analyzer. I already have it working.

 Buy 4 HackRF.

The analyzer works with a single hackRF.

I have tried it in two and it works very well, in the other 2 it does not show me the signal as in the others. I think you have to calibrate them but I do not know how.

I'm working on Raspberry, with QT, libhackrf. version 2018.01.01

Thank you very much.

dominicgs, 2018-04-23:
HackRF One is uncalibrated test equipment.  You *could* calibrate it, but the results aren't necessarily going to last for a long time as they vary with temperature and other factors.

Matioupi, 2018-08-02:
I have fitted my hackrf one with a nooelec tcxo. From hackrf_debug output, it seems to be detected.
However, when using kal (from https://github.com/scateu/kalibrate-hackrf ) I have output results that are way out tcxo specs and that vary a lot from test channel to test channel.
I don't know if this is a kal bug or hackrf issue... looks similar to this : https://github.com/scateu/kalibrate-hackrf/issues/6
The output are pretty consistant from run to run in same channel, but largely channel dependent.
Any idea why t would be that way ?

dominicgs, 2018-08-02:
Do you see the same issue when you run kal without the tcxo connected?

Matioupi, 2018-08-02:
I just tested without the TCXO and results are the same.
From https://github.com/scateu/kalibrate-hackrf/issues/2 it seems it may be related to software not updated when hackrf firmware went from unsigned to signed samples. But I've been trying to patch this in kal without success... There must be something else.
My guess is that the issue is rather at kal level than on hackrf.
Would be nice if someone can replicate this...

Matioupi, 2018-08-07:
Just for the record, i'm gettting much more consistant results with https://github.com/JiaoXianjun/LTE-Cell-Scanner

`DPX CID A      fc   freq-offset RXPWR C nRB P  PR CrystalCorrectionFactor
FDD  82 2 1852.5M          149h  -5.4 N  75 N 1/6 1.0000000804849560332
FDD  81 2 1852.5M          129h -8.42 N  75 N 1/6 1.0000000696863771132

FDD  82 2 1852.5M          136h  -4.5 N  75 N 1/6 1.0000000733270317799
FDD  81 2 1852.5M          126h -10.3 N  75 N 1/6 1.0000000680980221102

FDD  82 2 1852.5M          130h -4.43 N  75 N 1/6 1.0000000699071471821
FDD  81 2 1852.5M          126h -10.4 N  75 N 1/6 1.0000000679814959881

FDD  72 2 1835.1M          124h -3.59 N 100 N 1/6 1.000000067744795107

FDD 283 2   1870M          134h -3.67 N 100 N one 1.0000000716316523874
FDD 282 2   1870M          140h -10.5 N 100 N one 1.0000000747332455919

FDD  82 2    783M          101h -3.75 N  50 N 1/6 1.0000001285996318323
`

closed 2019-11-26 by miek