greatscottgadgets/hackrf #475: How to handle underflow/overflow with libhackrf?

awelkie, 2018-04-21:
Are the tx/rx callbacks expected to always return immediately or are they expected to block? For example should the tx callback block if more samples aren't yet available, or should it return with an error code, or should it return with some bogus data (all zeros, maybe) and no error code? And if the callbacks should block, should they block indefinitely or should they have some timeout?

dominicgs, 2018-04-23:
The callback is intended to be non-blocking.  If the callback blocks we'll get underruns and the HackRF will re-transmit previous buffers. I think you ask a good question about what to do when no more samples are ready.  It comes down to, what do you expect to be transmitted over the air in those situations? 

 If it's a packet based radio system and you reach the end of the packet without filling the buffer, packing it with zeros is appropriate.

If you've got a continuous transmission and can't produce samples quickly enough, then you've got a fundamental problem with your data rate.

One option could be to have the callback return the number of samples that it has produced, but that won't help when the data is produced at a lower rate than it's transmitted.

hustpigeon, 2018-05-02:
@dominicgs how about first packet， before we call device callback function, the transfer had been submitted by libusb_submit_transfer. that means every time we transmit data, the first packet is garbage data. The next tranfer,we copy data to transfer buffer,then the data is meanfull.

straithe, 2021-11-12:
It looks like @awelkie's questions was answered so I will close this issue. @hustpigeon if you still have your question, please open an new GitHub issue.

closed 2021-11-12 by straithe