greatscottgadgets/hackrf #476: how to make a OFDM  transceiver  with Hackrf One

littleRanger, 2018-04-26:
### Steps to reproduce
1. I am trying to tx and rx  OFDM signals using two hackrf one. 
2. I tried the tx&rx grc in gnuradio. /share/gnuradio/examples/digital/ofdm ，but I cannot receive any packet. error msg: 
"gr::log:INFO:packet_headerparser_b0-Detected an invalid packet at item 128864 "
"gr::log:INFO:header_payload_demux0-parser returned"
3. I dont know  what does it mean. and how to solve the problem.Thanks.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**: Ubuntu16.04

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2018-04-30:
Those look like informational messages rather than errors, and they come from the GNU Radio blocks that you're using.  They don't appear to me to be representing an issue with your HackRFs.

Unfortunately we don't have any experience with the GNU Radio OFDM example flowgraph.

I suggest asking on the gnuradio mailing list or in the #gnuradio IRC channel on freenode.

closed 2018-04-30 by dominicgs

reshma07mr, 2023-01-13:
Hi, is the issue resolved, if so how is it done ? 

Because I am also experiencing the same while trying to implement data transfer (Text or audio file) based on examples located in the folder gr-digital/examples/ ofdm. The signal is produced and transmitted, but not demodulated. The receiving file is 0 bytes in size.

I have reaised a ticket for the support on this under #1249