greatscottgadgets/hackrf #480: usb ACK what use usb_transfer_schedule_block()???

hustpigeon, 2018-05-01:
[image]

dominicgs, 2018-05-01:
This vendor request is used to write firmware to the SPI flash chip.  We send the firmware binary in chunks on the bulk out endpoint.

The function `usb_transfer_schedule_block()` is used to tell the USB stack where to write the data from the bulk out endpoint once it arrives.  Once the buffer has been copied this function is called again with the `stage` argument set to `USB_TRANSFER_STAGE_DATA` and we can then write the buffer to the chip.

hustpigeon, 2018-05-02:
@dominicgs  thank you very much， the other question, when we ACK **setup transaction**， is usb stack ack by default? the code shows we ACK  "data transacton" and  "status transaction", but not ack "setup transaction"

hustpigeon, 2018-05-02:
i get it ,the function usb_transfer_schedule_ack() is not the real function that ack usb data， it is only set usb stack to receive or send data. may be usb ack data by default?!!

hustpigeon, 2018-05-05:
the usb stack will send ack packet by defualt , or automatically, programer need not to care it.

closed 2018-05-05 by hustpigeon