greatscottgadgets/hackrf #481: hackrf_sweep use case failure

jestrada2006, 2018-05-04:
### Steps to reproduce

I apologize if my post is not in line with protocol but here are the steps to reproduce.

1.  Start PC, running Ubuntu 16.04 LTS
2.  Connect 2 HackRFs to PC that are running firmware version 2018.01.01
3.  Open Terminal with CTL+ALT+T
4.  Navigate to folder with the shell script written below.
5.  Execute shell script below

```

#!/bin/bash

while :
do
           count=0
           cmd1=”hackrf_sweep -a 0 -l 8 -g 8 -d a06063c82335505f -f 902:928 -B -w 100000 -r junk.bin”
           cmd2=”hackrf_sweep -a 0 -l 8 -g 8 -d a06063c824873a5f -f 2400:2500 -B -w 100000 -r junk1.bin”
           $cmd1 &
           cmd1_PID=$!
           $cmd2 &
           cmd2_PID=$!
           while [ $count -lt 60 ]
           do
                      sleep 1
                      count=$(( $count + 1 ))
           done
           kill -SIGINT $cmd1_PID
           kill -SIGINT $cmd2_PID
done

```

### Expected behaviour

The expectation is that hackrf_sweep will run for 60 seconds on both HackRFs simultaneously.  After 60 seconds, it should receive the SIGINT signal and close on both HackRFs.  Then, it should restart again and continue forever without issue.

### Actual behaviour

After three to four hours of normal operation, I receive the following message resulting in a file size of 0 bytes:

```

call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 902 MHz to 942 MHz
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_set_amp_enable(0)
Stop with Ctrl-C
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
call hackrf_set_amp_enable(0)
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 1.00018 seconds (0.00 sweeps/second)
79 total sweeps completed, 78.99 sweeps/second
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
161 total sweeps completed, 80.49 sweeps/second
242 total sweeps completed, 80.66 sweeps/second
324 total sweeps completed, 80.99 sweeps/second
406 total sweeps completed, 81.19 sweeps/second
...
...
...
```

As you can see, only one HackRF is scanning but this problem is not unique to a particular HackRF or USB port.   I have gone through all possible combinations of sets and my memory consumption is normal and the htop application does not show me any unusual signs of excessive CPU usage.  Also, it is not unique to the shell script because I have a program in python that it supposed to implement the same functionality but experiences the same error.

Note that this error does not occur when running hackrf_sweep continuously on its own.

Why is it doing this and what can I do to fix it?  Thank you!

dominicgs, 2018-05-04:
It looks like you're hitting a bug where the HackRF doesn't get reset to an idle state properly when the transfer thread is killed.  As you can see from your experimentation, it's not a very common bug, but I understand how frustrating it is and I'm trying to resolve it.

As a workaround, could you run hackrf_sweep continuously and parse the output data in to 60 second chunks?

jestrada2006, 2018-05-04:
The purpose of my experiment is to conduct real-time analysis on the results I obtain from hackrf_sweep.  This analysis is done on a different system entirely and is transferred over IP using a modified version of FTP.  Using the tools that you have available and without extensive edits to the hackrf_sweep.c, file I could not think of a way to transfer the file to this system without first stopping a sweep and then sending the file.  

I chose this method because I had assumed the hackrf_tools had been vetted and I did not want to disrupt their integrity.  If you have a better idea, I am open to it. Otherwise, I can't do anything else other than what I am doing.  

I appreciate your help in this.  The deadline for my project is approaching and if the solution is not readily available, I will probably have to find an alternative one.  Thanks again for your help!

dominicgs, 2018-05-15:
I think this is something that needs to be added to hackrf_sweep.  IT could probably do with having a number of sweeps or time limit value.

closed 2019-11-27 by miek