greatscottgadgets/hackrf #484: TCXO problem . Hack RF not found .

anthonynacz, 2018-05-08:
### Steps to reproduce
1. Connect a TCXO on the hackrf board
2. use hackrf_info command
3.

### Expected behaviour
Hack RF should be recognized for use .

### Actual behaviour
Receiving at the terminal following message : hackrf_info version : unknown
libhackrf version : unknown (0.5)
No hackRFs boards found

### Version information
**Operating system**: Ubuntu 17.10

**hackrf_info output:**
Receiving at the terminal following message : hackrf_info version : unknown
libhackrf version : unknown (0.5)
No hackRFs boards found

dominicgs, 2018-05-08:
Does it work correctly if you disconnect the TCXO?

Does anything appear in the output of dmesg when you connect the HackRF?

anthonynacz, 2018-05-09:
It does work correctly without the tcxo. 
When connecting tcxo i do not receive anything in dmesg

dominicgs, 2018-05-09:
It sounds like the tcxo could be causing a short or something similar.  Are you certain that it's installed correctly?

Another possible issue is that the firmware on the HackRF is not dealing with the presence of the new oscillator, but I've never heard of that happening before.  Could you give me the output of `hackrf_info` when the tcxo is not connected?

detjar, 2018-05-17:
Guys, I have similar behavior. @anthonynacz  did you solve your issue?

anthonynacz, 2018-05-17:
I flipped the tcxo 180 degrees and installed it in the same place. The issue was resolved

dominicgs, 2018-05-18:
Thank you both for letting us know what the issue was.  Hopefully this helps other users in the future.

I'm going to close this issue now, unless there is any problem outstanding?

detjar, 2018-05-18:
From my side all is OK now. The issue was there: [img]https://i.imgur.com/FSK0VOP.jpg[/img]

anthonynacz, 2018-05-19:
No problem with me as well

closed 2018-05-21 by dominicgs