greatscottgadgets/hackrf #485: Technical Questions using HackRF One

mehrdaddavoody, 2018-05-09:
1) When using HackRF One, is the any way we can extend the RF bandwidth only by software (GNU Radio Companion or other supported software) in order to see the FFT of different signals at a wider frequency domain? If yes, how much wider the RF bandwidth would be?

2) Is there any way to sense when a signal is transmitted? (Because of the fact that saving a captured signal into a file using "File Sink" creates a file with a large size depending on capturing time, I was wondering if there is a way to trigger the "file sink" only when a signal is being received).

3) Can we figure out a way to make a flow graph in GNU Radio or other HackRF supported software to make a "hopping central frequency" at different sets of frequencies (e.g. 100MHz, 150MHz and 200MHz) to sense a set of desired frequency bands with a scheduled hopping pattern? If yes, what are the timing limitations?

mehrdaddavoody, 2018-05-10:
@dominicgs Thank you very much for your kind reply. In fact, I am aware of this wonderful feature, but I don't quite get what I want using it. What I want is a feature that scans and saves some given frequency bands in milliseconds in a loop. 
Another issue I am challenging with right now is a feature in GNU Radio Companion to "trigger" the file sink block to save a received signal just when it's being received in order to minimize the file size. I know it's proper to ask this in GRC github page but I thought some people here might have gone through this already and might help me solve this issue.

dominicgs, 2018-05-10:
You could use a squelch block in GNU Radio to achieve that, or possibly the "burst tagging" block, but I've never used it, so I'm not certain on that one.

As for sampling multiple bands, are you saying that you'd like to record data from a band for a short time, then hop to another frequency and record samples there, and so on in a loop?  Rather than simply performing an FFT on that data?

mehrdaddavoody, 2018-05-11:
@dominicgs Consider a scenario in which you are expecting a signal with an unknown frequency and what you know is just some probable frequency bands in which the signal might pop up. In addition to it, these frequency bands are further than 100 MHz from each other. Since there is a 20MHz limitation on the bandwidth around the central frequency using HackRF One, is this even possible to detect this signal? Or our only way of detecting this signal is to count on our luck and scan these frequency bands as a loop, hoping the signal might occur within the band that was being scanned?

dominicgs, 2018-05-12:
Ah, now I think I understand.

Unfortunately, the answer is no, there's no way to capture all of those frequencies and capture the transmission with a single HackRF.  If the signal was repeating then I would recommend locating its frequency with hackrf_sweep and then capturing on that band. This would also work if the transmission was very long, although you'd miss the start of it.

If the bands are more than 100 MHz apart, you're unlikely to have much more luck with a single USRP.  In my opinion this is a job for multiple radios.  If the bandwidth is narrow enough, you could use a handful of RTL-SDR dongles.

I have to say, I'm curious about a transmission that happens only once and the transmit frequency isn't known in advance.

closed 2019-11-25 by miek