greatscottgadgets/hackrf #486: hackrf one transmit multiple channel?

haryows, 2018-05-15:
greeting sir,

i want ask a simple question actualy, did hackrf one can transmit multiple channel?
if yes, how can i did it with my hackrf. thank you sir.

Version information:
hackrf_info version: git-5e9cad6
libhackrf version: git-5e9cad6 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c834771ccf
Board ID Number: 2 (HackRF One)
Firmware Version: git-5e9cad6 (API:1.03)
Part ID Number: 0xa000cb3c 0x00584756

dominicgs, 2018-05-15:
This very much depends on what you mean by "channel".

HackRF can transmit 20 MHz of arbitrary signal.  If your signal is less than 10 MHz you can probably transmit multiple simultaneously, depending on channel spacing.

dominicgs, 2018-05-16:
No, in that use case you would need multiple radios or a radio with 300 MHz of bandwidth.  HackRF has 20 MHz of bandwidth.

haryows, 2018-05-16:
thank you very much for the answer sir.

closed 2018-05-16 by dominicgs