greatscottgadgets/hackrf #491: Suggestion for HackHF not HackRF version.

Injunear, 2018-05-29:
Hello, 

I got the HackRF One primarily to use as an inexpensive SDR Ham radio.

I would like to see a Hack**HF** version that covers as low as possible up to about 60 MHz (or at least 30). 

I expect that by limiting the frequencies so low the design might be able to better find and amplify the HF signals that hams and shortwave listeners are interested in.

Thank you.

mossmann, 2018-06-01:
How do you envision such a design improving upon existing designs for HF such as [KiwiSDR](https://www.seeedstudio.com/KiwiSDR-Kit-p-2725.html)?

Injunear, 2018-06-01:
Thank you very kindly for responding. I wasn't aware of KiwiSDR until your email.

My speculative thinking was that regular expensive ham SDRs would be able to pull better HF signals out of the noise than the HackRF. That was just a guess.

I was "assuming" that because of the wide reception range, signals in the HF range would not be as good as on an expensive ham SDR.

(Yes, I know the old saying about "assume". It might apply here. 😉 )

I suppose I should have confirmed that assumption before requesting a major redesign.  😊

I am completely green in this whole field so I guess what I want to know before asking about a Hack-HF is how well does the HackRF reception compare with the expensive ham SDRs?

If it is comparable then I with draw my question. I don't need the lower bands. I was just hoping for better better reception on the HF bands.

(I also probably ought to return the Ham-It-Up I bought based on the same assumptions.)

Thank you very much for your time.

P.S. I am going through your video tutorials about how to use the HackRF. They are helpful and I like the way you present them but I have a lot questions the videos don't cover. Do you know of other tutorials that explain how HackRF and GNURadio (or GQRX) work together?

Thank you again for your time.

mossmann, 2018-06-14:
The main things that expensive ham SDR platforms (like FlexRadio) give you that HackRF does not are better filtering, better amplification, and more dynamic range.  The first two of those can be addressed by adding external amplification and filtering to a HackRF.  The dynamic range is something that can't be improved without a major redesign, but for many applications the need for more dynamic range can be mitigated with adjustable gain and/or oversampling.  The one thing HackRF can never do as well as a platform with greater dynamic range is detection of a very weak signal that is adjacent in frequency to a very strong signal.

The Ham-It-Up has some nice filtering but no amplification.  I think you might be very happy with it if you also add an LNA to your signal chain.

I recommend the [GNU Radio Guided Tutorials](https://wiki.gnuradio.org/index.php/Guided_Tutorials) and various videos from the GNU Radio Conference.

Injunear, 2018-06-15:
Thank you very much for taking the time to respond.

I will look into a LNA. I have realized that the RTL-SDR that I have does not perform the way I want it to. Perhaps an LNA is what I need.

Thank you for your time.

closed 2019-11-25 by miek