greatscottgadgets/hackrf #492: How to measure Eb/N0 using HackRF?

H1Rong, 2018-06-04:
HackRF works as a receiver. How to measure the Eb/N0 of the received signal?
Thanks!

straithe, 2021-11-06:
Hi, @H1Rong do you still need help answering this question?

jerrydsp, 2021-11-10:
Greetings H1Rong,
I do not know much about the HackRF device as I'm still learning about it.

1. Calculating Eb/N0
 By definition (Eb/N0 is the Energy Per bit i.e Power of a single symbol (QPSK  signal) x Baud interval (Tb)
 divided by
Noise power spectral density i.e. Noise PSD

1.1 From the signal plot determine the power in a single baud and X baud rate
1.2 Calculate the noise power from the power spectrum and divided by the signal bandwidth ... N0

2.  Measuring the performance of the De-modulator.
 No short method, sadly.

Please note that a QPSK signal is actually 2 x BPSK signals whose two carrriers are in phase quadrature.

2.1 You need a variable amplitude/power noise source - white Gaussian noise - to add to your QPSK signal
2.2 Signal sources from 2 x pseudo-random binary sequence  (PRBS) generators
2.3 Generate 2 carriers in phase quadrature.
2.4 Multiply signal #1 by carrier and signal #2 by carrier 2 ... you now have two BPSK signals. Add these two BPSK signals to give you the QPSK test signal. Adding various amounts of noise will give you the test signal.

3. Demodulation
3.1 Extract the carrier.
3.1.1 A Costas loop (2 PLL's) will recover the two carriers - best solution
3.1.2 Squaring a PSK signal will also produce the carrier you then create the in phase and quadrature components (carrier1 and carrier2)
3.2 Multiply each carrier by the data stream and low pass  filter the output - this is bit stream you desire. For best results make the length of this LPF equal to the number of samples in a baud - this will me a filter matched to the signal - the best you can do !!
3.4 Compare bit output - use an exclusive OR gate - and count the errors -
 gate output  =1 no error
 gate output = 0 error

4. An very easy way to look at the signal will be to generate a scatter plot.
4.1 Shift your QPSK  signal  to base-band i.e. 0Hz. This is very easily done by generating a  of -f0 Hz. Where f0 is the signal centre frequency. This carrier will be a complex number i.e. S(t)* (cos(wt)+isin(wt))
4.2 Plot the real and imaginary components the less noise the smaller will be the spread.

That's it in words.

5. Carrier synchronisation is a critical part of this exercise.
5.1 Easiest is to square a BPSK signal and you will have the carrier - filter this signal to recover the carrier. 
5.2 A very narrowband filter increase the performance.

Sorry to be so long winded - but your questions are quite complex.

Jeremy.

straithe, 2021-11-15:
I am going to close this issue as I think @jerrydsp has answered this question very well.

closed 2021-11-15 by straithe

jerrydsp, 2021-11-15:
Please may I leave a last comment:

Measuring the Eb/N0 of a PSK signal is particularly easy. Just compute the Power Spectra Density (PSD) of the PSK signal, this has the form of (sin(theta) / theta ))**2 , so the Eb/N0 of this signal is the ratio of the peak of the main lobe (Eb) to the level at the first null (N0) - piece of cake ! eh?

straithe, 2021-11-15:
Haha. You are always welcome to leave more comments. Thank you for your very detailed initial response as well!

jerrydsp, 2021-11-15:
No problem - thanks.