greatscottgadgets/hackrf #494: HackRF-One as a standalone node.

HrfTdoa, 2018-06-11:
Interested to implement TDOA using several stand alone HackRF - One sdrs.
Each node will be synchronized with TCXO, send data over GSM(using UART), and be powered by a battery via micro USB.
RF DSP related logic is to be implemented on the Cortex-M4 micro-controller.

Several questions:

1. Was HackRF-One designed for IoT standalone applications? 
If so – was the intention to use USB for battery and UART for Cellular? 
Is there a more efficient set up?
2. Seems that Cortex-M4 micro-controller can be programmed to perform DSP related functionality, as demonstrated in:
https://www.aliexpress.com/item/New-HackRF-One-1-MHz-6-GHz-SDR-Platform-Software-Defined-Radio-Development-Board-Aluminum-alloy/32812930490.html

Looking for references of RF DSP algorithms implemented on HackRF as a standalone platform.
Any leads to related projects are welcomed.

Thanks in Advance.

mossmann, 2018-06-14:
HackRF One can be used as a stand-alone platform.  As an example, see the [PortaPack H1](https://github.com/sharebrained/portapack-hackrf).  PortaPack is typically powered by a USB battery.

closed 2018-06-14 by mossmann

HrfTdoa, 2018-06-16:
Thank you Michael.

In addition to being stand-alone platform, the device also should support a remote FW updating functionality.
Seems that for this purpose, a host will be needed. 
One possibility is to use Raspberry Pi.
This can also solve the DSP processing load, as raspberry pi 3 supports GNU radio.
My question is, can raspberry pi be avoided by using the existing HW of the HackRF.

Thanks.