greatscottgadgets/hackrf #495: NEED HELP/ADVICE ON TWO PRODUCTS (on eBay) FOR MY HACKRF

orrinwinton, 2018-06-13:
Hello folks, received my HackRF a week ago. I hope it's okay to ask about a couple products on eBay, to ask if they are appropriate or recommended for the HackRF:

(1) RF power meter, allegedly -75dBm to +15dBm. I would run the HackRF into this, with a 50-ohm dummy load on the output of the meter. Only goes up to 500 MHz i know: 
https://www.ebay.com/itm/1-500Mhz-OLED-RF-Power-Meter-75-15dBm-Power-Set-RF-Attenuation-Value-battery/132248196401?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2055119.m1438.l2649

(2) The external TCXO: is this the appropriate part to buy, and do the pins go nicely into a slot inside the HackRF. (It looks like this prevents the case from closing? Are there TCXOs available that can be connected to the SMA port on the rear of the HackRF?):
https://www.ebay.com/itm/NEW-external-TCXO-clock-PPM-0-1-for-HackRF-one-GPS-Applications-GSM-WCDMA-LTE/182507445522?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2055119.m1438.l2649

I would like to put "labels" on this post, like "help needed" and "question" but i can't find the labels. ... Thanks for any comments on this. AD0NR & the group de WN1Z Orrin.

unixpunk, 2018-06-13:
http://www.nooelec.com/store/tiny-tcxo.html - I came across this one the other day, fits all BUT the original plastic case.  I have the TCXO you linked to but haven't actually used it yet.  I can tell you that it does NOT fit the original plastic or the aluminum case.  HTH  (Can't comment on the power meter)

ly2ss, 2018-06-13:
1. For power meter I'd look for this:
https://www.aliexpress.com/item/8GHz-1-8000Mhz-OLED-RF-Power-Meter-55-till-5-dBm-Sofware-RF-Attenuation-Value-D28/32848991165.html

But I can't say how good is it (calibrated or not)
I chose to make myself one, based on ad8313 module (look f.e. DL2SBA power meter) and calibrate it with known signal source.
2. I bought large TCXO module but I had to do some work to squeeze the module in metallic box for HackRF One (Hammond 1455J1201 case)
https://www.aliexpress.com/item/HackRF-one-external-TCXO-clock-PPM-0-1-for-GPS-Applications-GSM-WCDMA-LTE/32709992757.html

chudgoo, 2018-06-14:
Since we're talking about it... Is there a TCXO option for Portapack users?

The portapack attaches to those pins, but I can't tell if it actually USES them...

ContextPerspective, 2018-06-14:
If it helps those looking for TCXO + finding a case which actually fits....here is what I use:

- https://www.ebay.com/itm/NEW-external-TCXO-clock-PPM-0-1-for-HackRF-one-GPS-Applications-GSM-WCDMA-LTE/182507445522
- https://www.ebay.com/itm/Transparent-Acrylic-Sheet-Housing-Case-For-HackRF-One/111977984102

Both of these work together without issue.

orrinwinton, 2018-06-14:
Thanks for comments. LY2SS i will probably try that meter, at that price. With a 20dB attenuator it could cover up to +15dB etc.

Info on the TCXO & cases: very helpful. Do folks think the following TCXO listed on Amazon will fit into the silkscreened(?) plastic case (the case has 'HackRF One Great Scott Gadgets' printed on case):
https://www.amazon.com/dp/B077H7Q7GL/?coliid=I3A4RVGHBFSQ6J&colid=3UEP8D3IXMCMZ&psc=0&ref_=lv_ov_lig_dp_it

They sell an aluminum case they say *will* accommodate the TCXO.

Since i think i'm going to buy the PortaPack, i may hold off on the TCXO. Which brings up a related topic (sorry if i'm wandering off topic here): the idea of getting one of the cheap HackRFs from China ($170) and putting the TCXO on that. (Because the PortaPack will go 'mobile' with no TCXO.)

I bought my HackRF off Amazon (NooElec i think) at full price. ... I wonder if the ones from China are any good. If so, which ones. Would not want to get an earlier hardware version.

Thanks again for comments. I was blown away by the hackrf_transfer function, makes sense though. Lots of studying to do, to do well at constructing flowcharts that *work* for NBFM and AM.  ==orrin

unixpunk, 2018-06-14:
Check nooelec site, that TCXO says will not fit original great scott gadgets case, only their acrylic or aluminum.  I have the aluminum and can confirm the ebay tcxo does NOT fit in it.

I also have a portapack, no txco, easy to adjust ppm on the fly.

closed 2018-06-21 by orrinwinton