greatscottgadgets/hackrf #496: Pentoo .iso Which one ?

ProperTricky, 2018-06-21:
Firstly i feel embarressed even to ask but trying to set up my HackRF I'm stumped at the first hurdle.
 When on the mirrors page to download the Pentoo .iso instead of clicking the link to download the full package i get individual files so which 1 do i need and then from the sub menu ? 
[image]

barto-, 2018-06-21:
The file `README_which_version_do_I_want.txt` explains the different choices.

```Pentoo comes in many flavors and it is important to choose wisely.

Right now, you have two main choices:

Choice 1: hardened or default

You want hardened.  No seriously, you want hardened.  When was the last time
you thought to yourself "I need less security in my pen-testing environment?"
In all seriousness, nearly everything works in the hardened builds, and it is
vastly more stable than anything you have ever used before with the added bonus
of being more secure.  You only want default if you are doing exploit
against yourself, or you *need* opengl support.  OpenCL and CUDA work fine in
the hardened release, but right now, opengl support still eludes us.  If you 
cannot live without opengl acceleration pick default, otherwise, you really
want hardened.

Choice 2: x86_64 or i686

Reality is, no one tests things on x86 anymore.  I don't just mean us, I mean
software devs, distro devs, and even kernel devs.  No one cares about old
hardware.  If you download the i686 builds, it will run on anything Pentium M
or higher, but reality is that the x86_64 builds are not only faster, but
hugely better tested. Pick x86_64 unless you have a really good reason (like
wanting to support ancient hardware).```

ProperTricky, 2018-06-21:
Yes ive seen that so i want hardened, but which Hardened and then which 1 from the submenu as i originally asked ...

barto-, 2018-06-21:
If you have a relatively easy machine:
http://mirror.switch.ch/ftp/mirror/pentoo/latest-iso-symlinks/pentoo-amd64-hardened.iso

ProperTricky, 2018-06-21:
You wonderfull person. Thank You.

ProperTricky, 2018-06-21:
So ive downlaoded and burned to cd, ive booted and run from the cd and put in a password which has been accepted but it's stuck on pentoo@pentoo ~ $

ProperTricky, 2018-06-22:
Each time i try to install from a flash drive i get "can't find root block device" , so reading the web ive found i have to change the kernel line script ? ... Will that fix it ?

dominicgs, 2018-06-26:
If you're normally a Windows 10 user, I would probably recommend not using Pentoo, but using the GNU Radio live image or Ubuntu - they're much simpler to get started with when switching from Windows.

It's also possible to get most/some of the tools in binary form for Windows.  The latest build from [here](http://downloads.myriadrf.org/builds/PothosSDR/) probably contains enough to get you started.

rbeede, 2018-08-24:
No more readme in the download directory now.

What is the difference between Full and the others?

necrose99, 2018-08-28:
@Sabayon Linux is easy for most , duual boots well esp if laptop has 2nd drive bay. 
has a pentesting enman overlay.   made from pentoo , however can add much of pentoo into sabayon via layman...  normally i'd recommend pure pentoo , but the UEFI in my older ROG ASUS is a touch flakey..  
best of all much of any other Gentoo overlays are available..  and can add Pentoo default-amd64 as binrepo...

closed 2018-12-01 by dominicgs

bill-e-ghote, 2018-12-16:
As it happens, I run an old-AF-x86 system, only  because I'm a hoarder and won't let it go. Until recently, I was even able to build chromium, despite only having 2G RAM. I might be the only person anywhere building Gentoo on x86.