greatscottgadgets/hackrf #499: HackRF at 5.8 GHz

hoyunwon, 2018-07-02:
We are currently building some system at 5.8 GHz. Through VNA, we found that S11 parameter at 5.8 GHz is about -4dB. At 2.4 GHz, it was showing near to -11 dB, which I wanted. Is there any way I can realize high return loss at 5.8 GHz?

lukeweston, 2018-07-04:
This doesn't sound like a HackRF issue.

So you're transmitting power into the HackRF port, and you're finding that S11 is higher than you desire.
The general answer to that question is that the impedance of the load should be matched to the impedance of the source. For example a 50 ohm load should be connected to a 50 ohm source. More correctly, the impedances are complex numbers, and the load impedance should be the conjugate of the source impedance, for example if a RF SoC has a port impedance of 40+7j ohms then the antenna (or antenna plus matching network) should have an impedance of 40-7j ohms looking into its port.

If you need to transform the impedance on one half of this system, there are options like pi networks, transformers or other types of matching networks. But you need to start by specifying, or measuring, the impedance looking into both ports.

mossmann, 2018-07-12:
I believe this question is regarding Opera Cake, not HackRF One, but I don't really have an answer to the question.  We're interested in improving the high frequency performance, but we haven't found a way to do so yet.  It is likely that the finished product will perform similarly to current prototypes.  One thing you can do is to ensure that the correct PCB substrate and stack-up are used in manufacturing and to request that the PCB is tested for controlled impedance.  A 12 mil trace on the top layer should have a characteristic impedance of 50 ohms.

closed 2018-07-12 by mossmann