greatscottgadgets/hackrf #500: HackRF freeze in transmission (TX) with computers in USB3 and not with one with USB2

fe8769a, 2018-07-11:
### Steps to reproduce
1.hackRF freeze in transmission (TX) with computers in USB3 and not with one with USB2
2.
3.

### Expected behaviour
We are using hackrf with the tetrapol-kit decode projet 👍 

https://brmlab.cz/project/sdr/tetrapol/audio

With the following line 

./tetrapol_tx.py -i ./transmit.bits

### Actual behaviour
On a USB2 computer, hackrf transmits the information then switch to RX , re switch to TX .
On USB3 computers ( tested on 4 differents computers ) , the hackrf stay in TX mode . Sometime after 3-5 minutes, sometime after one hour . Files sent are always the same .
Tests has been made connecting a usb2 hub to a USB3 port and the hackrf freezes

In order to restart , we have to stop the TX program , unplug the hack, replug the hackrf and restart the TX until the next freeze

### Version information
Generally OS is ubuntu . mine is Xubuntu 15 . Hard drive is the same from computer in USB2 and USB3 
remove SSD in USB2 and put it in USB3 computer 

**hackrf_info output:**
Firmware is release 2017.02.1

WE didn't find a linux command that could make a reset (unplug) of the hackrf that we could insert in our transmit batch 

Any comments are welcome 
```

dominicgs, 2018-07-13:
When the HackRF stays in TX mode, is it transmitting a signal? Repeating the end of the previous signal?  Just a carrier?

Would it be possible to let us know the USB controller models? (lspci output?)

dominicgs, 2018-07-14:
Have you tried this natively on a host with USB3 ports?  This sounds like it could be an issue with vmware.

Does the HackRF respond in this state?  Canyou run `hackrf_info` or other tools?

fe8769a, 2018-07-15:
I did tests with native USB3 laptop . Hard disk is the same on USB2 laptop (working fine) and USB3 LAPTOP ;
Hackrf stay in transmit mode but not in GSMK modulation as it should be but with a strange fixed noise 

Hackrf_info says hackrf is busy .

The Only way to restart is to unplug and re plug hackrf . 

using a usb2 Hub on usb3 port has the same problem

fe8769a, 2018-07-15:
I switch my Core I7 laptop in USB2 in BIOS and I haven't any problem any more . It is the way hackrf or xubuntu are managing USB3 wich is wrong . How I don't know

dominicgs, 2018-07-16:
Thank you for taking the time to test and confirm that.  I will investigate it.

Is it possible for you to try the host code and firmware from git rather that the release?  If not, there will be a release in afew weeks time, which may contain a fix for this.

fe8769a, 2018-07-16:
I ll wait for the next firmware release that will correct that . I confirm that my laptop in usb2 Mode on the usb3 port is working fine . Let me know when firmware will be available

ckoval7, 2019-07-06:
I am having a very similar/the same issue. Sometimes the HackRF locks up after a few TX cycles, sometimes it takes a while. I don't have any USB 2.0 ports to test with, so I can't say whether or not switching to 2.0 fixes it.

martinling, 2022-03-18:
I think this is probably fixed as of #1029, which dealt with several bugs that could cause HackRF to get stuck when repeatedly starting & stopping TX or RX, with symptoms as described.

All of those bugs involved race conditions, and due to the timing dependencies, some were only reproducible on certain machines. I suspect that the USB3/USB2 differences reported here are similarly coincidental.

The strange signal when it got stuck in TX would have been the old buffer contents repeating. That will no longer happen since #982.

I'm going to close this because I'm fairly sure it should be resolved now. If anyone is still able to reproduce a problem with the current master branch, please open a new issue.

closed 2022-03-18 by martinling

fe8769a, 2022-03-18:
Hello
Please update to last Hackrf firmware . It solve this USB3 trouble