greatscottgadgets/hackrf #501: Newbie Question

KevinSDR, 2018-07-17:
Hello there, 

So I have an ethernet connection between two computers, one of which is connected to a HackRF as a host and the other is not connected to any SDR but simply has the source file to be transmitted via ethernet. the third computer is connected to another hackRF:

comp1 ===ethernet===>comp2 +hackRF====wireless===>comp3+hackrf
(source file)
 What I need to do is the following:
1- Receive data coming from Comp1 and decode it at the hackRF+Comp2.
2- re-encode and modulate the recieved data to transmit it wirelessly at some frequency Y to hackRF+comp3.

If some one could just turn me to the right  direction to implement the first part of the scenario ( proper way to receive and decode the received data from the ethernet connection), I would be grateful.

Thank you,

Kevin

### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

unixpunk, 2018-07-17:
Hi Kevin, the options here are better suited to this question: https://github.com/mossmann/hackrf/wiki/Getting-Help as are the forums on rtl-sdr.com.  To get you started, have a look at sdrangel and sdrdaemon.  Those two will allow what you're looking to do.  Comp1 would run sdrangel w/the sdrdaemon source plugin and comp2 would run sdrdaemon server with the hackrf libs.  HTH

dominicgs, 2018-07-19:
Hi Kevin,

Are you using GNU Radio?  I recommend taking a look at the UDP source and sink blocks.

closed 2018-10-01 by dominicgs