greatscottgadgets/hackrf #502: new cpld programming

TonyHagen, 2018-07-20:
i experienced a static charge from the antenna and then later on i found the hackrf stopped working.
Considering if i need to change the CPLD - COULD YOU PLEASE GUIDE ME how to program the new CPLD and the tools i require it

unixpunk, 2018-07-20:
https://github.com/mossmann/hackrf/wiki/Updating-Firmware  For future usage-related questions please also see these options for better places to get assistance: https://github.com/mossmann/hackrf/wiki/Getting-Help  Hope that helps!

cjheath, 2018-11-06:
A charge from the antenna is quite unlikely to zap the CPLD. In my zapped unit, U13 (the front-end LNA MMIC) is blown. What symptoms are you seeing that make you suspect the CPLD?

TonyHagen, 2018-11-06:
Agree, the CPLD was a hunch – YES NOTHING went wrong to CPLD its safe. THE front end 2 MMIC RF amplifier is blow – changed and its now working.

I raised an issue till the CPLD in apprehension if I need to burn the firmware what is the way if someone is now aware of UBUNTU/UNIL and is familiar with only WINDOWS

73
Tony

From: Clifford Heath 
Sent: Tuesday, November 06, 2018 12:03 PM
To: mossmann/hackrf 
Cc: TonyHagen ; Author 
Subject: Re: [mossmann/hackrf] new cpld programming (#502)

A charge from the antenna is quite unlikely to zap the CPLD. In my zapped unit, U13 (the front-end LNA MMIC) is blown. What symptoms are you seeing that make you suspect the CPLD?

cjheath, 2018-11-06:
See my recent issue for a candidate protection device for the MMIC. It should be fitted on the RX_AMP_IN line, either side of C58 - but needs a series capacitor because both sides of C58 are above earth. That's what I plan to do anyhow. This MMIC is effectively directly connected to the antenna input with no protection, and that's just madness.

cjheath, 2018-11-06:
Also, when you say the front end 2 MMICs are blown, I don't know what you mean. The next MMIC is quite a long way from the first one. Unless you're transmitting, U25 is not really in the circuit, and even then, it's the output not the input. Which was the 2nd MMIC you replaced?

cjheath, 2018-11-06:
Assuming you're talking about the HackRF One, there's only one LNA. The other nearby one is connected to the input switch (U13, SKY133) and is the transmit amp U25 - but the *output* is connected not the input so it's unlikely to blow from an input pulse. The third path from U13 bypasses the LNA to U12, which goes to U9 to switch filters in or out of circuit. There's not another MMIC there.

If you replaced U25, are you sure you needed to?

cjheath, 2018-11-06:
The TVS clamps at +-15V, which is useless to protect anything here, especially the MMICs. The SKY13317 is rated to pass almost a watt, so it's not very susceptible. So the input of that poor MMIC is basically directly connected to the antenna with no protection.

TonyHagen, 2018-11-07:
Yes, both were short – so we need to replace

73
Tony

From: Clifford Heath 
Sent: Tuesday, November 06, 2018 5:11 PM
To: mossmann/hackrf 
Cc: TonyHagen ; Author 
Subject: Re: [mossmann/hackrf] new cpld programming (#502)

Assuming you're talking about the HackRF One, there's only one LNA. The other nearby one is connected to the input switch (U13, SKY133) and is the transmit amp U25 - but the output is connected not the input so it's unlikely to blow from an input pulse. The third path from U13 bypasses the LNA to U12, which goes to U9 to switch filters in or out of circuit. There's not another MMIC there.

If you replaced U25, are you sure you needed to?

cjheath, 2018-11-07:
Ok, thanks, that changes my plan to add protection. I didn't realise that the Tx MMIC would be vulnerable. I think now I'll add the protection in parallel with the TVS, not at the input of the receive LNA.

cjheath, 2018-11-08:
I intend to use Skyworks SMP1330. It's a part that colleagues have had success with in some very difficult protection scenarios (sub-uV receivers that must also correctly receive multiple volts of direct RF). It will do the job.

miek, 2019-11-27:
As far as I can see the original issue was solved so I'm gonna close this. If you're still having any trouble with an amp failure please do open up a new issue, thanks.

closed 2019-11-27 by miek

cjheath, 2019-11-27:
Does this mean we can expect some action on the second-oldest outstanding issue, https://github.com/mossmann/hackrf/issues/541 ?

My offer to integrate the required changes into a current version of the PCB is still open.