greatscottgadgets/hackrf #505: set_freq_explicit / channel inversion bug

gibbi, 2018-07-25:
### Expected behaviour / Actual behaviour
When we look at the firmware (tuning.c) the q channel is inverted for every frequency above 2150MHz. If i want to use set if/mixer explicitly instead the channel is only inverted if the MixerLO is higher than the IF. I assume both paths (explicit and auto) should work the same?

This bug makes it very hard to use explicit settings at all.

Did i miss something?

hustpigeon, 2018-12-05:
did you solved this problem?

straithe, 2021-01-26:
Are you still experiencing this issue?

gibbi, 2021-01-27:
i did not check in a long time.. got it working well enough without the use of explicit tuning

straithe, 2021-01-27:
Do you need any further assistance or may I close this ticket?

straithe, 2021-02-11:
I'm going to close this for now as it seems your issue has been solved, but please re-open this issue or open a new one if you still need assistance.

closed 2021-02-11 by straithe