greatscottgadgets/hackrf #506: Signal Strength Misrepresented in qspectrumanalyzer for 0GHz-2.1GHz

phlogistic, 2018-07-27:
Seen this behavior in multiple different physical locations. Same laptop, shielded / ferrite core monoprice USB cable, same hackRF in different locations.

[image]

[image]

### Steps to reproduce
1. Install qspectrumanalyzer
2. Update firmware on hackRF to 2nd to latest firmware release, if necessary
3. Use hackRF to sweep 20MHz-6GHz
4. See 20MHz-2.1GHz is misrepresented for signal strength
5. Confirm by going to multiple different physical locations

### Expected behaviour
20MHz-2.1GHz signal strength is not a plateau

### Actual behaviour
20MHz-2.1GHz signal strength is a plateau

### Version information
**Operating system**:

Ubuntu 16.04.1

**hackrf_info output:**

TBD

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

qspectrumanalyzer 2.1.0

dominicgs, 2018-08-01:
What sort of antenna do you have connected to your HackRF?

While that step is very abrupt, it's at a point where we change RF paths and we would expect there to be some change in signal strength because the different RF paths have different characteristics.

I'm more concerned that there appear to be no distinct signals showing up anywhere in the entire sweep.  I would normally expect there to be some signals showing and the noise floor between -60 and -70 dB.  However, you appear to have very specific levels in the colour levels chart on the right.

phlogistic, 2018-08-08:
ANT500

konikofi, 2018-08-09:
Looks like ANT500 is rated for 75MHz to 1GHz.
https://www.adafruit.com/product/3584

dominicgs, 2018-08-17:
@phlogistic As we discussed in person at defcon, the ~3 dB drop at the frequency where we switch RF paths is not unexpected.

However, the completely flat performance across the rest of the frequency range suggests an issue.  Can you try running this again with a different antenna?  Even a piece of wire or an unfurled paperclip would do for testing.

closed 2018-11-30 by dominicgs