greatscottgadgets/hackrf #509: Suggested LNA/PA replacement model number

cyrus104, 2018-08-07:
I have recently diagnosed my kickstarted HackRF One with a back PA and LNA.... I'm sure it was my fault.

I have been looking at mouser and digikey but cannot find the exact model anywhere besides maybe china in stock. While I don't mind buying parts from china, it normally take weeks to get to me.

My question is do you know of a replacement for the  Avago MGA-81563 that is available? 

Thanks

unixpunk, 2018-08-07:
My LNA is toast as well, PA still good though.  I bought both replacements a while back but forgot which is which, check the Wiki here too, there is a better place to ask questions like this than here in the issues section.  I don't know if these are the 'perfect' replacements, I think I found them listed here: http://www.t4f.org/articles/repairing-the-hackrf/

http://d.digikey.com/R00N1wnXg7000070S9K4qp4
http://d.digikey.com/b00047XN4Sa0K7nr000gw1p
HTH

cyrus104, 2018-08-09:
Thanks for your response, however I've been using the same t4f.org site for replacing the LNA/PA. If you didn't notice the first line for the LNA/PA on digikey in your post has nothing available. If you trying an order it from them they will email you a few hours later saying they won't be getting anymore. I did the same thing with mouser as well, I called them first about it.

I have a few MGA-81563 ordered from ebay but would like to find an in production replacement.

unixpunk, 2018-08-09:
Ah interesting, I bought mine a while back, had no trouble then.  There should be a replacement part with same/similar qualities available, I would think.  Sorry I can't help further.

lukeweston, 2018-08-10:
Perhaps the Hittite/Analog Devices HMC311 is OK.
Same package/pinout, but the noise figure is a little bit higher and
changes to the bias may be needed.

mossmann, 2018-08-22:
I recommend continuing to use MGA-81563. It is still available from some distributors:
https://www.findchips.com/search/MGA-81563

GSG is happy to replace or repair any HackRF One we sold (including via Kickstarter). Contact the reseller you purchased it from or contact us at info@greatscottgadgets.com if it was a Kickstarter unit.

closed 2018-08-22 by mossmann

unixpunk, 2018-08-22:
@mossmann Thanks for the tip!  I emailed info@ a while back and they said GSG doesn't do repairs.  I have no way of knowing if the bad LNA was my fault or not, I assume it was, so I'm not looking to get repairs for free.  :)

mossmann, 2018-08-22:
It is true that we do not normally do repairs, but we definitely do replacements.

NickHead, 2018-08-27:
Just do what I am going to do, forget about it and go back to using RTL-SDR, least you can buy them cheap enough. I just can't believe that you cannot get spares easily for such a well advertised bit of expensive kit and I am sure as hell not sending it back for repair when I could easily do it myself, feeling a bit let down really.

NickHead, 2018-08-28:
I have already repaired it once, can't remember where I got the part from but I know it cost me dearly. I don't over load the input, that I know of , There have been a lot of people with the same problem , so , why is there not a supply of LNA repair kits ? Satellite systems had PSU repair kits because they had a known fault. I just think we are in a situation if it works it's OK, if it goes wrong, you are on your own.
I for one would not recommend it to anyone, not as though my opinion will count for anything as i have bought one and now on my own.

MadCamel, 2020-02-29:
I managed to find some MGA-81563 chips, but there was a minimum order. So I have quite a few left after repairing my hackrf. I'd be willing to drop a few in the mail for free to anyone with a blown tx/rx amp. Contact me on Freenode if interested.

sv5byr, 2020-04-16:
Hello dear friend,
I toast my HackRF PA too and I need MGA-81563 if you have any available.Thanks in advance.

Sasszem, 2024-11-10:
I just repaired a defective clone device. I could not find reputable distributors for the MGA amp, and aliexpress orders were defective / fake parts.
I found that it is possible to replace it with a MiniCircuits [`ERA-2SM+`](https://www.minicircuits.com/WebStore/dashboard.html?model=ERA-2SM%2B) amp. It has very similar electrical properties, but has better transient and ESD protection, and hopefuly will last longer. 
The package is NOT pin compatible, and requires some careful bodging - the new amp can be fitted rotated. It also requires a small resistor for biasing, I used 10 Ohms in series with the L12 RF choke. I see the latest revision also added one, but the existing devices will need that modification.
Also, the choke got destroyed by soldering, so I replaced it with a new one (that has an order of magnitude higher SRF), and that could effect my results as well.
I plan to present some measurements later.