greatscottgadgets/hackrf #511: current (bleeding edge) firmware souce does not produce a workng firmware

bl4krat, 2018-08-10:
My HackRF One was shipped with the latest release firmware (2008.01.1). Both the original firmware and the firmware found in firmware-bin for the latest release work correctly (in that 'hackrf_transfer -r /dev/null -s 20e6' produces a stream of data from the hackRF at around 40MiB/sec).

However, the current version of the firmware (as of github today - last update appears to be 5 months ago) builds without errors, but results in 'hackrf_transfer -r /dev/null -s 20e6' producing a stream of data from the hackRF at around 20MiB/sec. Its as if the data is one byte per sample, rather than the expected 2. The example FM receiver built in lesson 1 runs but you only hear a nasty raspy version of static. data is dropped even at 10MHz sample rate.

(FWIT, I can compile the latest release firmware (2018.01.1) from source and it works fine)

```
root@kali:~/Develop/hackrf/firmware/hackrf_usb/build# hackrf_transfer -r /dev/null
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
10.0 MiB / 1.000 sec = 10.0 MiB/second
10.0 MiB / 1.000 sec = 10.0 MiB/second
10.0 MiB / 1.000 sec = 10.0 MiB/second
10.0 MiB / 1.000 sec = 10.0 MiB/second
^CCaught signal 2
 0.8 MiB / 0.062 sec = 12.7 MiB/second

Exiting...
Total time: 4.06255 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

root@kali:~/Develop/hackrf/firmware/hackrf_usb/build# hackrf_transfer -r /dev/null -s 20e6
call hackrf_set_sample_rate(20000000 Hz/20.000 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
19.9 MiB / 1.000 sec = 19.9 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
19.9 MiB / 1.000 sec = 19.9 MiB/second
^CCaught signal 2
 3.9 MiB / 0.184 sec = 21.3 MiB/second

Exiting...
Total time: 3.18507 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

```
### Steps to reproduce
starting with a fresh kali install (which includes packages 'hackrf' and 'libhackrf0'), with gnuradio installed and also cmake and gcc-arm-none-eabi:
1. git clone https://github.com/mossmann/hackrf
2. add the necessary libopencm3 submodule, following the instructions.
3. build hackrf-usb,bin following the instructions
4. Upload with hackrf_spiflash
5. press reset
6. hackrf_transfer -r /dev/null -s 20e6

### Expected behaviour
The data transfer rate should be approx twice the sample rate as happens with  the 2018.01.1 version of firmware.

### Actual behaviour
The data transfer rate is approx equal to the sample rate at any tested sample rate

### Version information
**Operating system**
Kali Linux (Weekly build W32 2018)
pc: Dell Inspiron 15-5577 'gaming' laptop (Intel Core i7-7700hq, usb ports are all USB3, nvidea GTX-1050m graphics)

**hackrf_info output:**
```
root@kali:~/Develop/hackrf/firmware/hackrf_usb/build# hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
[... 648 lines cut ...]
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006d4766
root@kali:~/Documents/GSG-SDRwithHackRF_tutorial/firmware/latestRelease/hackrf-2018.01.1/firmware/hackrf_usb/build# hackrf_transfer -r dev/null -s 20e6
Failed to open file: dev/null
root@kali:~/Documents/GSG-SDRwithHackRF_tutorial/firmware/latestRelease/hackrf-2018.01.1/firmware/hackrf_usb/build# hackrf_transfer -r /dev/null -s 20e6
call hackrf_set_sample_rate(20000000 Hz/20.000 MHz)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
^CCaught signal 2
17.6 MiB / 0.440 sec = 39.9 MiB/second

Exiting...
Total time: 4.44113 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

```

dominicgs, 2018-08-10:
The firmware changes in git require a CPLD update to go with them.

`hackrf_cpldjtag -x hackrf/firmware/cpld/sgpio_if/default.xsvf`

bl4krat, 2018-08-10:
Yep, That's got it. 

Thanks.

closed 2018-08-10 by bl4krat