greatscottgadgets/hackrf #512: Hackrf stuck in upgrade firmware

haryows, 2018-08-15:
### Steps to reproduce
1. Download the fimrware from https://github.com/mossmann/hackrf/releases/tag/v2018.01.1
2. Extract file firmware>firmware-bin, n run " hackrf_spiflash -w hackrf_one_usb.bin"
3. Stuck

### Expected behaviour
I expect will get succes notification or do next instruction.

### Actual behaviour
I just stuck in this : 
[image]

I stuck in here for an hour n nothing happen

### Version information
Pentoo 2018.0

**hackrf_info output:**
[image]

what I should do for this stuck?i have no idea on it. help me please.

haryows, 2018-08-20:
thanks for ur suggest sir, very appreciate that.
i'll try and report here as soon as posible.

Slasher006, 2018-12-27:
i have found out that it works if you load the dfu from the release package and then write the bin from master. With the one that master builds it hangs at spi flash.

closed 2019-11-26 by miek