greatscottgadgets/hackrf #517: issue after firmware update

devel505, 2018-08-26:
### Steps to reproduce
1.hackrf_sweep
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 1.00052 seconds (0.00 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
2. i changed out 4 usb cables used tried updating a few times , and used three diffrent computers and a rasberi pi
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

GravesJake, 2018-08-27:
Which firmware version did you update to?

devel505, 2018-08-27:
the latest on 2018.01.1 also when I press dfu reboot the same issue happens and if I use DFU command line it doesn't do anything

GravesJake, 2018-08-27:
Can you post the output of 'hackrf_info'?

devel505, 2018-08-28:
hackrf_info
hackrf_info version: git-1ae7c21
libhackrf version: git-1ae7c21 (0.5)
Found HackRF
Index: 0
Serial number: 00000000000000009258
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00654761

GravesJake, 2018-08-28:
There's a version mismatch between your host code and your firmware. Your host code is the latest from git and your firmware is from the release tools. Either rollback your host code using the release tools or update your firmware to the latest git version.

closed 2019-11-26 by miek