greatscottgadgets/hackrf #518: HackRF Hats

necrose99, 2018-08-28:
RPI3 or Rock64/Pine64 over gpio. 

curious , as I can use the ports of Pentoo Linux to arm64 
(VPS) I could start to publish...  my own dropbox is starting to fill up.  :-(  but for a good cause. 

point being is could make for an all in one case..  with hackrf /extra wifi snooping  in a more compact arrangement...

mossmann, 2018-08-29:
I don't understand any of this.  I'll close it if there isn't clarification.

necrose99, 2018-08-29:
http://www.hamblog.co.uk/top-10-amateur-radio-uses-for-raspberry-pi/ 
RASPBERRY PI / **PI HATS & BONNETS** & ADD-ONS https://www.adafruit.com/category/286

could just put RPI3 + Hackrf on RPI3 or[ Rock64](https://www.pine64.org/?page_id=7147) as main cpu.. 

ie ALL in one box ..  

dint know if you had plans or board kits in the PiHAT or bonnet format pulling power / cpu etc from RPI... etc.. 

**CASE+IOT+HACKRF ...  all in one KIT..**  

forgive sleepy at the moment , _was a brainstorm of an idea.._   earlier , Was just asking  if that might be in the works..   been a long day.     

anycase been researching using IoT devices in a swarm to augment pentesting ie small cheep / Force multiplication... task distribution...  just having all in a box kits... makes for easier to hide on engagements...

RPI3 / Rock64 has GPIO pins , and slam a hackRF "HAT" on those perhaps was the main idea...