greatscottgadgets/hackrf #519: Broken RF Amp

spaghet, 2018-08-30:
I'm not sure if this is the right place for hardware issues, but i'll try.

I ordered a genuine GSG HackRF off of eBay, It has worked fantastically for the price I got it for ($150) other than needing a few firmware updates. Recently, I noticed a severe drop in TX power, after a little investigation I discovered that turning the RF amp off causes the power to recover. This leads me to believe that my RF amp has gone bad.

Anyway, I was wondering if there was somewhere I could send it to get fixed as I don't think I have the necessary tools/skill to fix it myself. Thanks!

mossmann, 2018-08-31:
I agree with your diagnosis. We'll replace or fix it for you. Send it to us with a note that mentions issue #519, and be sure to include your return address.
https://greatscottgadgets.com/contact/

closed 2018-08-31 by mossmann

mossmann, 2018-09-01:
Yes, please. There is a good chance that we will replace your unit instead of repairing it. Send it either in the original enclosure or just the PCB. Actually the bare PCB would be best because it will remind us to replace your shield cage if that is soldered on.

spaghet, 2018-09-01:
Okay, Thanks!