greatscottgadgets/hackrf #522: generate signal in the wrong frequency

weitinglu, 2018-09-12:
Recently I get a hackrf, I am try to use DFS Pulse Tester. But I facing some problem.
I not sure it is my hackrf's HW fault or software not setting in the correct way.
It could generate rf signal but in the wrong frequency.
I had issued this problem in DFS Pulse Tester, But I don't get a response.

DFS Pulse Tester:
https://github.com/adriangranados/dfs-pulse-tester

My system is using ubuntu 18.04.
install driver/software by pybombs.
everything is up-to-date.

hackrf_info :
hackrf_info version: git-1ae7c21e6c
libhackrf version: git-1ae7c21e6c (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000644064dc243c72cd
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00744758

GNU Radio Companion 3.7.12.0

My friend helps me with using the spectrum to get the result. The result as follow.

Wi-Fi DFS Channel | Spec | Measured | Delta
-- | -- | -- | --
52 | 5260 | 5486 | 226
56 | 5280 | 5520 | 240
60 | 5300 | 5555 | 255
64 | 5320 | 5592 | 272
100 | 5500 | 5912 | 412

How to check my hackrf is working perfectly? is It there a tool or script that can generate the signal I chose?

dominicgs, 2018-09-12:
You can use `hackrf_transfer` to generate a carrier at a chosen frequency.

`hackrf_transfer -c 127 -f 5260000000` will generate a carrier at 5.26 GHz.

closed 2019-11-26 by miek

markl17, 2020-06-29:
hi I have a frequency counter and it showing 3184 mhm when I type hackrf_transfer -c 120 -f 52600000 in Mac
it should be showing 52 mhz.something but it's totally off on the screen in display of hackrf it says 52600000 so I don't know what to believe my oscilloscope is telling me that its putting out 25 mhm when I put out 1000 hz in gnu radio so I think the unit is missing a screw