greatscottgadgets/hackrf #523: HackRf not being seen by computer

joe91lew, 2018-09-19:
My hackRf is getting power but not being seen by my computer. The reset button works, but the DFu button has zero response. I have tried 3 cables. 3 computers. With no changes. That being said my other HackRf was good to go

dominicgs, 2018-09-19:
When you say the 'DFU button has zero response" do you mean pressing it alone does nothing, or pressing it as you reset or power on the device does nothing?

Is this a brand new HackRF? Or have you been using it successfully until now?

joe91lew, 2018-09-19:
Both. The DFU has no response at all. The reset button works fine. But whether I hold the reset or not.  The DFU does not work. 

Joe

dominicgs, 2018-09-19:
Is this a new HackRF?

joe91lew, 2018-09-20:
I am trying to update the firmware on both of my hackrf ones. They seem to
have gone out on us. Wondering the best method of doing so. I have been
trying to update one of them on linux (ubuntu) but the directions are
confusing me. Also when i type command "lsusb" I am getting "ifc9:000c, but
its still not working.

dominicgs, 2018-09-27:
There are two methods for updating firmware

1) `hackrf_spiflash` - if you have a working HackRF then you can update the firmware using `hackrf_spiflash -w hackrf_one_usb.bin`, having obtained the file `hackrf_one_usb.bin` from a release package.

2) If you are unable to connect to the HackRF you can use DFU mode.  In DFU mode the HackRF will appear with a USB ID of `1fc9:000c`.  You can update it with `dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu`.  The HackRF will then appear to the host system and you can update it using the above method.

Adewale121, 2018-10-15:
hello pls who can help me out ......in what range could Hackrf work i was told it's 100 meters 
and if that's the truth which other devices can i use to have a hugerange

dominicgs, 2018-11-30:
@joe91lew Did you manage to update your firmware successfully?

Adewale121, 2018-11-30:
Yes bro

miek, 2019-11-26:
I'm going to close this as there's not been a response in a while, but if you're still having issues feel free to re-open or open a new issue, thanks!

closed 2019-11-26 by miek