greatscottgadgets/hackrf #526: Building the hackrf-tools on wondows 10 with cmake

mgbx, 2018-09-29:
### Steps to reproduce
1. building hackrf tools
2. building libhackrf
3.

### Expected behaviour
Tell us what you expect should happen
I was expecting the cmake configuration and build to go smoothly
### Actual behaviour
Could NOT find LIBUSB (missing: LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR)
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Threads (missing: THREADS_PTHREADS_WIN32_LIBRARY
  THREADS_PTHREADS_INCLUDE_DIR)
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  ../cmake/modules/FindThreads.cmake:242 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:49 (find_package)

### Version information
**Operating system**:
Windows 10 Home OS version: 17134.286
**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
C:\hackrf-master\host>cmake libhackrf
-- Building for: Visual Studio 15 2017
-- The C compiler identification is MSVC 19.15.26730.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.15.26726/bin/Hostx86/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Could NOT find LIBUSB (missing: LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR)
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Threads (missing: THREADS_PTHREADS_WIN32_LIBRARY
  THREADS_PTHREADS_INCLUDE_DIR)
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  ../cmake/modules/FindThreads.cmake:242 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:49 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/hackrf-master/host/CMakeFiles/CMakeOutput.log".

dominicgs, 2018-10-01:
When building on Windows you need to explicitly direct cmake to the directories where it can find libUSB and pthreads.

I have updated the [instructions on the wiki](https://github.com/mossmann/hackrf/wiki/Operating-System-Tips).

There are also pre-compiled binaries available as part of the PothosSDR [packages](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D)

James-Brown, 2018-10-01:
This might help you..
http://www.seti.net/notebook/volume-10/Chap66/chap66.php

regards...... Jim

From: Dominic Spill 
Sent: Monday, October 1, 2018 10:14
To: mossmann/hackrf 
Cc: Subscribed 
Subject: Re: [mossmann/hackrf] Building the hackrf-tools on wondows 10 with cmake (#526)

When building on Windows you need to explicitly direct cmake to the directories where it can find libUSB and pthreads.

I have updated the instructions on the wiki.

There are also pre-compiled binaries available as part of the PothosSDR packages

mgbx, 2018-10-02:
Okey, after installing the PhotosSDR packages I can now use my HackRF tools, but when i run hackrf_info, this is the output:

C:\Program Files\PothosSDR\bin>hackrf_info
hackrf_info version: git-1ae7c21
libhackrf version: git-1ae7c21 (0.5)
Found HackRF
Index: 0
hackrf_open() failed: Access denied (insufficient permissions) (-1000)

Any clue how I can get access to the HackRF?

dominicgs, 2018-10-02:
This looks like a permission issue.  Could you try running the same from a command prompt that you're started with the "run as administrator" option?

mgbx, 2018-10-03:
Thank you. That helped, and it seems to be working fine now!

closed 2018-10-03 by dominicgs