greatscottgadgets/hackrf #528: Add Pseudo-Doppler support to `hackrf_operacake` tool

dominicgs, 2018-10-01:
Depends on #527 

The current pseudo-doppler mode is triggered by a special combination of opera cake port settings.  We should have a new USB API call to start and stop PD mode.

This could be an extra call from an existing tool or a new command line tool.

The tools should be documented in the way that other tools are documented on the wiki.

williamg42, 2019-03-26:
I've been working on a tool an have already merged docprofsky's firmware and the most recent hackrf master but I have seen no sign of special opera cake port settings in the firmware. However calling the appropriate USB APIs do not seem to start the antennas feeds rotating. Any information on that special combination of opera cake port settings would be helpful.

Nzbu, 2021-06-03:
> 
> 
> I've been working on a tool an have already merged docprofsky's firmware and the most recent hackrf master but I have seen no sign of special opera cake port settings in the firmware. However calling the appropriate USB APIs do not seem to start the antennas feeds rotating. Any information on that special combination of opera cake port settings would be helpful.
> 
> For anyone looking for information:
> You need to set PA and PB to 0, in the operacake_set_ports(uint8_t address, uint8_t PA, uint8_t PB) command

How did you set the PA and PB ports to 0? I've tried with the "hackrf_operacake -o 24 -a 0 -b 0", but it does not seem to start the switching. I've also merged these codes.

williamg42, 2021-06-03:
If you check my fork I had to hard code it during the hackrf startup
sequence. I was not able to get the USB commands to work either.

Nzbu, 2021-06-03:
Thank you! Is the version of the code on your fork functioning, so I could try it?

williamg42, 2021-06-04:
It should be. I don't remember what I set the "rotational rate" but you can
find that in the main code.

maxroseGIT, 2021-06-14:
Hi @williamg42, thanks for the reply on @Nzbu question.

Can you be a bit more specific: What is the 'main-code' from your point of view?

And where did you hard code PA and PB to get the antenna switching started?

Thanks for letting us know.

williamg42, 2021-06-14:
Sorry, I guess I should have phrased that better. The command is placed in
the main function in hackrf_usb.c.

maxroseGIT, 2021-06-14:
Hey @williamg42, thanks for clarification. Can you give us some hint how much time you spend figuring out all required changes to make the switching work? Did you have contact to the developers?

For me it seems quite a challenge to dig through the code and find the important places of code that make the opera-cake working like I want it to...

So I am happy if you could point out where else (perhaps filenames) you did modifications to the original firmware. Or which could be also quite a nice thing: Some identifier-comment to the source code where you put modifications. With that you make the days for the followers (at least me and @Nzbu) a lot easier to customize your hard-coding for our needs.

I know that you're probably working on other projects right now, but if you find some time for rough documentation of you changes... splendid :) And huge thanks

williamg42, 2021-06-14:
Besides merging the branches the only change I had to make was adding the
function call to the main function of the program. All functions of the
opera cake are documented in the operacake c and h files and are fairly
straightforward.

Best of luck!

mossmann, 2021-12-11:
closed in #931

closed 2021-12-11 by mossmann