greatscottgadgets/hackrf #529: Pseudo-Doppler Documentation

dominicgs, 2018-10-01:
Write a wiki page to explain how Pseudo-Doppler works with HackRF One and Opera Cake.

The page should:
 * describe the technique or point to resources that do
 * explain how H1 + OC is sample synchronous and what that gains users
 * give practical examples of how to use the tools
 * explain some of the limitations of the tool to answer likely support questions
 * it would also be good to include an example flow graph

mossmann, 2018-10-01:
depend on #527 and #528

mossmann, 2021-12-11:
This is partially implemented in #1017 which documents time mode. Documenting example usage of time mode for pseudo-doppler direction finding would still be nice.

peter-stricher, 2023-05-28:
Do you have now a good documentation on how to do pseudo doppler with the H1 and OC