greatscottgadgets/hackrf #530: Update Opera Cake documentation

dominicgs, 2018-10-01:
Update the description and documentation on https://github.com/mossmann/hackrf/wiki/Opera-Cake to reflect the new state of the hardware and software after we have finished the development work in #527 and #528

maxroseGIT, 2021-03-24:
Hey, I am really happy to tell you that I have just soldered an opera cake board and from first tests it seems to work fine. Thanks for the work on that project.

I miss a bit of information on the antenna switching, especially on the time based switching. I know that you guys plan to update the documentation here. Wanted to let you know that there is at least one user who is highly interested in using the antenna switch with time based switching!

So far the only little hint on how to trigger any antenna switching in [#528](https://github.com/mossmann/hackrf/issues/528#issuecomment-476862535) provided one has merged [docprofsky's](https://github.com/docprofsky) branch with the master branch.

Thumbs Up for further progress here! If I can help out somehow I would be glad to contribute somehow..

maxroseGIT, 2021-11-19:
Thanks to the efforts in #931 time based switching works now. At least in the way described in #931. When using the commands suggested in #931 the LEDs light up according to the selected ports A1 to A4.

I have some issues with the LED indicators next to the antenna ports. And I am not sure whether this is related to the electronics (which I soldered myself) or the firmware.

What is the expected behaviour of the LEDs on the Opera Cake Board in general?

straithe, 2021-11-29:
Can you open a new GitHub issue with your questions, please?

I'll be updating the documentation over the next while as Great Scott Gadgets ramps up Opera Cake production. So hopefully you'll see the documentation you want to see soon!

PR #1017 (2021-12-12): update Opera Cake documentation
closes #530
closed 2021-12-12 by straithe