greatscottgadgets/hackrf #536: HackRF RX/TX Problem

0xCoto, 2018-10-08:
Hi everyone,

I got a Bias Tee, and I connected the wrong end to the HackRF. I accidentally connected RF+DC to the HackRF and RF to the Antenna (the Bias Tee takes external power at 12V). The RF LED lit up (even though the HackRF was not connected to the computer), and after realizing my mistake, I connected the HackRF to my computer without any bias tee or anything to ensure everything is fine, it makes a bleep (indicating USB device got connected) and another bleep half a second later (indicating USB device got disconnected).
I tried to undo this by doing the same mistake, and this time I tried again, but now when I connect it to my computer, it does not detect it at all, and some noise is coming from the silver plate on the HackRF (this is how it sounds: https://cdn.discordapp.com/attachments/265199259291222016/490276346946387980/video.mov).

This is my Bias Tee: https://www.ebay.com/itm/RF-Divider-Biaser-Bias-Tee-10MHz-6GHz-for-HAM-radio-RTL-SDR-LNA-Amplifier-50VDC/113170701206?epid=21021845279&hash=item1a597fdb96:g:9Q0AAOSwvmxbWqtq

After explaining the problem to my uncle (who knows about electronics), he managed to fix the power problem by attaching some more cables: https://stargazerslounge.com/uploads/monthly_2018_09/5679E488-0622-42BF-A459-DF81680652B2.jpeg.3b3be22fbb6c785ba5834ce6e0bfc689.jpeg

Now everything works OK and the LEDs are properly lighting up. When I try to receive/transmit, the RX and TX LEDs light up accordingly just fine. However, it's completely deaf when receiving. No radio stations when there should be plenty - https://i.gyazo.com/7e4a4cee522f2ce9bd50b3500dda7098.png (issue persists at all frequencies)). Same thing for transmit, it just doesn't transmit at all (checked and listened with another receiver but nothing).

Could you give any predictions on what could have gone wrong here? I looked at the schematics (https://github.com/mossmann/hackrf/blob/master/doc/hardware/hackrf-one-schematic.pdf) and looking at the last image on the bottom right corner, I checked to see where the 12V could have gone to, going through the ANTENNA port.
First of all, the schematic seems incomplete (or at least I misinterpret it). How did the RF LED light up when I applied voltage through the ANTENNA port, even though there is no RF LED connection before C64? From what I understand, the 12V couldn't have gone through C64, so the damage should have happened on the MOSFET. I took out the 3-legged SP 4 chip, examined it with my electronics equipment, but it seemed OK. I tried RX'ing again without it, but the performance is identically problematic.

Let me know if you have any ideas!

Thanks!

mtoan2111, 2018-11-13:
I also got the same problem. Unfortunately, my HackRF has died.

cjheath, 2018-11-16:
If you fed 12V into the antenna, that will turn on the Bias T MOSFET and feed 12V into the 3.2V rail that almost everything runs off. I'm surprised you didn't get fireworks, but you're unlikely to be able to resurrect the HackRF.

mossmann, 2018-11-19:
I agree with @cjheath.  It is likely that that every IC on the board was subjected to 12 V and could have been damaged.

closed 2018-11-19 by mossmann