greatscottgadgets/hackrf #537: TX problem

mehdi-tazi-chellal, 2018-10-11:
Hi everyone,
I am new in using HackRF One. I am trying to generate a signal with GNU Radio( WINDOWS version) and transmit it with HackRF One. The problem is when I try to test if I transmit something with a second HackRF One or an spectrum analyser I don't receive anything and the TX led is red. I tried a lot of tutorials in the net but no one works. If anyone can help me, please do it because I am really blocked. Thank you!!

mehdi-tazi-chellal, 2018-10-12:
Hi,
Thank you for answering me. Now and after changing some parameters in sending and receiving part (RF,IF and BB gain), I can see the signal. But the quality of the signal changes when I change the band frequency( 2.4GHZ band, 5GHz band) and also It's a bad quality. If you know why it changes and how could I ameliorate the quality of the received signal please let me know. I joined here the flowgraph and the information in the command prompt. Thank you very much !!
[image]
[image]

dominicgs, 2018-10-12:
I have a few questions:
 * How are you defining quality of the signal here?
 * 1MSps is a low sample rate for HackRF One, could you try using a higher one?
 * What antennas are you using?  Or are the HackRFs connected through an attenuator?
 * Your signal source and osmocom sink are using different sample rates
 * The bandwidth of your GUI sink is 20 MHz but you're only feeding it 1MHz of data - this probably won't cause an issue, but it might look unexpected

mehdi-tazi-chellal, 2018-10-12:
Hi , 
when I am speaking about the quality of signal that's mean that I don't receive quietly the same emitted signal. There is lots of perturbations.
I don't use any antenna, I am using an SMA male plug. 
when I try to increase the sps I lose the signal. With this values I can see the signal but not the same one emitted.

mehdi-tazi-chellal, 2018-10-15:
Hello, 
No the HackRF are not diirectly connected. here is a photo of them.
[image]
Thank you :)

mehdi-tazi-chellal, 2018-10-22:
hello,
I am generating a signal using the vector source and I want to variate the values in the vector source with the block GUI RANGE. the problem is when I begin the simulation and I change the values nothing happen to the graph. Please answer me if you have any suggestions. here is a photo of my program in GNU RADIO. Thank you.
[image]

mossmann, 2018-11-19:
I'm seeing several different errors and misunderstandings represented in the above GNU Radio flowgraphs.  I suggest starting with my video series to learn how to use GNU Radio with HackRF:

https://greatscottgadgets.com/sdr/

closed 2018-11-19 by mossmann