greatscottgadgets/hackrf #538: Hackrf + RTL2832U for GSM

xcolacho, 2018-10-15:
Hi!
Sorry for not writing the template. It's just a question.
I'm a SDR newbie and I would like to make a rogue gsm bts. I know that GSM is full duplex. And right now I have a hackrf and an RTL2832U.
Can I use the two devices in one host to do it? If someone can guide me I would appreciate it very much. 

Thank you!

studysession, 2018-10-16:
...following...

cjheath, 2018-11-06:
It seems most unlikely that (using the HackRF as Tx and RTL2832U as Rx) you can get them to synchronise closely enough to work as a BTS. The SDR signal chain has quite a few delays in the processing pipeline, probably too much to cobble around and still meet GSM timeslots.