greatscottgadgets/hackrf #540: HackRF leds has are not being illuminated when is connected and Computer doesnt recognize the device

Jsanch32, 2018-10-29:
### Steps to reproduce
1. Connect Hack RF to the computer

### Expected behaviour
Hack RF leds are illuminated when is connected and Computer recognize the device

### Actual behaviour
Hack RF leds has are not being illuminated when is connected and Computer doesnt recognize the device

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2018-10-29:
Have you tried this with multiple USB cables?  Multiple ports?  and if possible, multiple host systems?

It sounds like the board isn't getting power if none of the LEDs come on at all.

Could you try holding down the DFU button as you connect it?

Jsanch32, 2018-10-29:
Hi Dominic,

I tried to use the HackRF with different USB cables from anothers HackRF’s , in different ports and holding down the DFU button and the HackRF is not illuminating the LED’s and its not being recognized by the computer. What it’s the next step for this issue?

Thanks & Regards, ☺

Jsanch32, 2018-10-29:
> Have you tried this with multiple USB cables? Multiple ports? and if possible, multiple host systems?
> 
> It sounds like the board isn't getting power if none of the LEDs come on at all.
> 
> Could you try holding down the DFU button as you connect it?

Hi Dominic, 

I tried to use the HackRF with different USB cables from anothers HackRF’s , in different ports and holding down the DFU button and the HackRF is not illuminating the LED’s and its not being recognized by the computer. What it’s the next step for this issue?

Thanks & Regards, :)

dominicgs, 2018-10-29:
It sounds like a hardware issue.  Please contact the vendor that sold you the HackRF and request a replacement.

Jsanch32, 2018-10-29:
Hi,

I’m contacting you because the solution from Great Scott Gadgets is to request a replacement. How we can proceed  with this?
I am looking forward to your answer.

Thanks & Regards, ☺

Jose Salvador Sanchez Cortes
System Validation Engineer
[cid:image001.png@01D377E2.57258FB0]
Visteon Technology Center Queretaro
jsanch32@visteon.com<mailto:jsanch32@visteon.com>
[cid:image002.png@01D377E2.57258FB0]
[twitter]<https://twitter.com/visteon>[you]<http://www.youtube.com/user/visteon?feature=results_main>[blog]<http://blog.visteon.com/>[link]<http://www.linkedin.com/company/2865?trk=tyah>[goo]<https://plus.google.com/+visteon#+visteon/posts>[fb]<https://www.facebook.com/VisteonCorporation>[cid:image009.png@01D377E2.57258FB0]<http://visteon.com/index.html>

dominicgs, 2018-10-29:
You need to contact the place where you bought the HackRF, rather than Great Scott Gadgets.

Jsanch32, 2018-10-29:
Yes I copy your email in Cc, just to let them know that You are aware of this.

Thanks ☺

Jsanch32, 2018-10-30:
Good morning Jose
I will send this request to returns.  They will get a hold of you to let you know what needs to be done.

Dawn Cassise Client Care Specialist
[Marco]<http://www.marconet.com/>

313.982-8995 , ext 1 | dawn.cassise@marconet.com
15260 Commerce Dr. S., Dearborn, MI 48120 | marconet.com<http://www.marconet.com/>