greatscottgadgets/hackrf #541: Antenna port needs better protection

cjheath, 2018-10-30:
### Steps to reproduce
1. Expose the antenna to nearby 17kW airport radar (under 2km distance) while in receive mode
2. Discover that the MMIC input amplifier is fried and receive sensitivity is greatly reduced

### Expected behaviour
The HashRF input should survive overload.

### Actual behaviour
The TVS protection device clamps at 15V, which is grossly inadequate. The SKY13317 can pass almost a watt, so the next device in the chain (MGA-81563 MMIC) will see the full 15V blasting through, which is more than enough to fry it.

At least, that's what we're surmising has happened. The HackRF (which I purchased in this damaged state) still receives, but with greatly reduced sensitivity. New parts are on the way to effect a repair, so I'll let you know how we go.

The input needs to have a solid RF clamp diode for proper protection, such as a Skyworks SMP1330. Also on its way...

### Version information
HackRF One 23 Feb 2014.

### Output
... very little.... :)

virendera, 2018-11-09:
Hey cjheath, I am also facing this issue, I already damage my two amp. It is possible to save it using Skyworks SMP1330. If yes then please tell me.

virendera, 2018-11-10:
Yes, I replaced amp and now it working fine on both Rx and TX but I want to protect it for future damage because I know it will be damage in near future. Can you elaborate the process to place SMP1330 on hackrf PCB. Are you know that someone already use SMP1330 with hackrf one and also it performance after placing it. I don't know how a cheap dongle  protect themselves from static charge. However for your reference I am just a software guy not electronic expert

virendera, 2018-11-11:
Ok, when you place it then inform me.

cjheath, 2018-11-15:
I traced a block diagram of the HackRF One front end, to see how far a damaging pulse could get.

The switches are no protection, because they will pass almost a watt, which is 20V p-p at 50 ohms. The TVS clamps this to +-15V, but the MMIC has absolute maximum (damage) levels of +0.5V, -1.0V at its input, and +6.0V, -6.0V at its output, which explains why these get blown up.

Assuming that the image filters offer some protection (which is unlikely, actually) the attached image has a red dot anywhere that could get zapped from the antenna. Potential problems go back to U17, the MAX2837, which is far too far. There are too many paths to protect them individually, so I'm going to fit PIN clamp diodes to the Vaa&GND rails right at the antenna, and back-to-back diodes at the input of the LNA U13. This should offer quite good protection. I'll add the P_1dB and IP3 calculations in a further post.

[image]

cjheath, 2018-11-15:
I hope that @mossmann will comment on the above plan?

cjheath, 2018-11-15:
Hmmm. There doesn't seem to be any clean 5V on the board, just Vbus, which we can assume is noisy and so don't want it anywhere near the Tx amp. If I clamp the Tx output to Vaa instead, that would limit the output to about 4dBm. Not great. Might need to series 2 more diodes to Vaa.

cardre, 2018-11-15:
Just a thought, the HackRF supports bias tee power, so there should be a ‘clean’ 3.3V source nearby enough to the antenna port?

Is 3.3V enough for what you’re trying to do?

I realise as a protection circuit, you wouldn’t want to turn this on/off by dis/enabling bias tee power, more that it just may be a supply that could be used nearby to tap into.

As per https://github.com/mossmann/hackrf/wiki/HackRF-One <https://github.com/mossmann/hackrf/wiki/HackRF-One> it can apparently support max 50 mA @ 3.3V.

Thought # 2, while again not ideal, maybe an external protection device (the circuit that you describe?) that plugs into the antenna port that could be powered by the bias tee power could be an option for a self contained unit for those that can’t/don’t want to modify their HackRF directly? Of course this would require them to enable the bias tee power to get the protection.

Both of the above assume 3.3V @ 50 mA is enough for what you’re trying to do, otherwise, as you were! :-)

cjheath, 2018-11-15:
Hello @cardre. No my plan above wouldn't work, and neither would yours, even though there's unswitched 3.2V nearby. The Tx amp output has a DC level at Vaa (3.2V) because of the inductor, and swings above and below by 1.7V.

I will need to couple it via a capacitor to a point biassed half-way between Vaa and GND, and clamp that point to not fall below GND-0.5 or rise above Vaa+0.5. That'll work, but it's a cap, two resistors and the double-diode.

And yes, an external circuit like this would protect the Tx amp output, but there's still a significant risk of overloading the Rx amp (input!).

mossmann, 2018-11-19:
@cjheath: The maximum input power is specified as -5 dBm.  See: https://github.com/mossmann/hackrf/wiki/HackRF-One#receive-power

The TVS diode is for the purpose of electrostatic discharge protection, not for protection from RF power.

SMP1330 is a nice suggestion for future designs, but I don't see a way that it would fit well into the HackRF One design.  I suggest using an external limiter.

mossmann, 2018-11-19:
@cjheath Certainly. If we can find a way to make it work, I may be able to get it into the next hardware revision, although it is a bit late for that: https://github.com/mossmann/hackrf/tree/h1r5

I've already replaced MGA-81563 partly because it has been more prone to damage than it ought to be and partly because it has been discontinued.

Is there a reason that you are considering a circuit more complicated than a traditional limiter circuit (e.g. as found in http://www.skyworksinc.com/uploads/documents/BRO372_11B.pdf)?

cjheath, 2018-11-20:
That PDF shows only a single-ended limiter, but it would need to be double-ended. It also shows limiting after a receive filter (that might sink some power when the clamp activates), which is not the case here.

The Tx amp maximum output level is well above the Rx amp's maximum allowable input level. If a user has no need for max Tx output, protection could entail simply changing the TVS for the PIN diodes. If anyone wishes to use the maximum Tx level, the clamping should allow 3.4Vp-p, so clamping it (via a coupling capacitor and equal resister divider) to GND and Vaa seems sensible. This can be done right at the antenna and should protect every possible signal path, except the Rx amp which must be clamped about +-0.5V for safety. The Rx amp input is at 0V, so can be easily clamped by another SMP1330 to ground with no other components needed - a tighter clamp than the one at the antenna input.

Even if it's too late to get this onto the new production revision, perhaps you could still add a footprint in time, so buyers can add their own protection?

What device did you choose to replace the MGA-81563? Did you find a suitable one that works below 100MHz this time?

cjheath, 2018-11-20:
Hmmm. The TRF37D73 is even more susceptible to damage than the MGA-81563 (10dBm vs 13dBm max P_in) and because it's leadless, cannot be replaced using a soldering iron. Please add the protection!

mossmann, 2018-11-20:
You seem to be basing your "even more susceptible" conclusion solely on the absolute maximum ratings of the parts.  In the case of the MGA-81563, however, that is invalid.  It is much more susceptible to damage than the absolute max rating suggests.

Why do you think a double-ended limiter is required (as opposed to single-ended)?

cjheath, 2018-11-20:
On the other hand, what makes you think the data sheets cannot be trusted?

The MGA-81563 has this, which says Absolute Max Range of RF input voltage to ground "+0.5 to -1.0". Notice why a double-ended clamp is needed? Contrary to the "maximum input power of 13dBm, the +0.5V peak figure corresponds to 5.5dBm into 50 ohms, a very low level - yet you have an antenna hanging directly off it (the switches offer zero protection). No wonder it's "susceptible to damage"!

[image]

The TRF37D73 has this, 10dBm maximum input power. It may also be assymetrical, meaning that its maximum input voltage is even less than 0.5V:

[image]

The TRF output Vmax is only 3.6V compared to the MGA's 6V also - it is a significantly more fragile device.

mossmann, 2018-11-20:
I think the MGA-81563 data sheet is lacking information about conditions that may damage the part.  For example, have you computed the power it would have been subjected to 2 km away from that 17 kW airport radar?

cjheath, 2018-11-20:
The airport radar is irrelevant, just a surmisal. Here is the data:
https://web.acma.gov.au/rrl/site_search.site_lookup?pSITE_ID=3684
https://web.acma.gov.au/rrl/site_search.site_lookup?pSITE_ID=3663
https://drive.google.com/open?id=1_0FDsSg6ieFXqkBVAonnSRmMBoV8oLYb
E.g. https://web.acma.gov.au/rrl/assignment_search.lookup?pEFL_ID=1035892
The EIRP on 2.85GHz is 43MW, and the distance is 2.3km. I'm sure you can do the math.

Your decision process mystifies me. People choose to put these parts into satellite missions, or not, based on the data sheet. What other evidence is admissible?

Even if the parts are *even more fragile* than the data sheet says, surely that's all the more reason add protection?

I don't understand your resistance to the idea. The SMP1330 would almost certainly have protected this device in this situation, without any detriment to the performance of the HackRF.

mossmann, 2018-11-23:
I'm resistant to the idea of adding SMP1330 to r5 because:

- I don't see room in the layout for it without significantly disrupting the entire RF section.
- I haven't thought of or seen a proposal for a specific and complete solution based on the SMP1330 that would provide useful protection for r5.
- r5 is already being prototyped, and it is probably too late to make this change for the first r5 production.
- Although MGA-81563 failures are more common than any other hardware failure, we've seen the problem in less than 1% of units sold.

I'm certainly open to incorporating RF limiting in a future revision.

cjheath, 2018-11-24:
Ok, I have multiple responses to this.

1) I don't think you're seeing all the failures. People blame themselves, and either stop using the RF amp, use an external one, or replace it themselves. In most cases, they complain about it somewhere and recommend others not to buy HackRF - I've seen a number of posts like this, folk who just stopped using it after changing the MMIC four times. How many people have bought something else because of reading these complaints?

2) Would you solder the MMIC without grounding your soldering iron, or using anti-static safeguards? No, didn't think so... yet this MMIC is directly connected to an antenna. Crazy. I'm sure people blow it by changing the antenna while it's powered (so the RF switch is on), because they don't realise that's incredibly dangerous. It doesn't need to be so dangerous, sorry, but it's just an irresponsible design. You have my total respect for everything else, and for making it open source, but not for this. And you're about to make it worse with an even more sensitive MMIC.

3) Looking at the layout, it seems that there's plenty of room if a couple of things get nudged over a little. SMP1330 is a very small device, and the MMIC needs no other parts to protect it. Few more parts to protect everything else, but they're not so sensitive, and there's more room near the antenna entry point.

4) There is a specific and complete solution for protection (involving two SMP1330's) in this very thread, including P_1db and IP3 calculations. What more do you want?

Seriously mate, add the protection. It's just good design, and good business.

virendera, 2018-11-25:
I agree with @cjheath, in my case I buy a  hackrf clone (Chinese) and I replaced it amp twice because I am not aware of static charge of anteena and transmitted without proper antenna connecting to it. Now, I learned some etiquette when using hackrf because I upgraded myself from rtlsdr to hackrf one and don't know these things. Without amp hackrf is deaf. Without amp protection all antenna work should be done after powering off to hackrf.

mossmann, 2018-12-06:
@cjheath It would be very helpful to me if you would draw a circuit diagram of your proposal.

I am having trouble understanding how a solution with a 1 dB compression point of 10 dBm would do much to protect a part (TRF37D73) with absolute maximum input power of 10 dBm.

On the other hand, I'm willing to consider changing the amplifier to a different part that can handle higher input power.  (I've been testing BGB741L7ESD.)  In that case, I can see how a limiter based on SMP1330 could be very effective.

cjheath, 2018-12-06:
Firstly, to protect the HackRF (except for the LNA), this circuit could be right at the input before the first switch (the two diodes are e.g. those in the SMP1330 package). My [comment](https://github.com/mossmann/hackrf/issues/541#issuecomment-438893489) shows that it won't limit the Tx amp power. After I calculated that, I realised that many other parts of the HackRF can be directly exposed to the antenna input (depending on switch settings) and would be protected by this same circuit. All those have lower levels than the maximum Tx amp output, so it would impose no additional limit.
[image]

cjheath, 2019-01-09:
My LNA is now protected. The SMP1330 is upside-down at an angle. It's a bit ugly, but for RF, if it's pretty, it won't work :) and this is as short as I could make it. The coupling capacitor C58 pretended to be a grasshopper, so I replaced it with an 0805 version.

If C55 was moved (it could be almost anywhere and work fine) there would be room for the SMP1330 with hardly anything else being moved.

[image]

I don't think I'll bother with the protection for the rest of the RF parts (the virtual ground sketched above) but if I was going to leave it connected to an external antenna, I'd definitely do that also.

virendera, 2019-01-09:
Giid work @cjheath :smile:

Would you explain to me in layman terms which new parts to use(bom) and how to place it in PCB ( layout).

cjheath, 2019-01-09:
I used the SMP1330, as discussed above. It's a SOT-23 package (two pins one side, one the other). The two on one side get connected together (little bit of wire). Then I connected those to ground (on the outer end of C55), and the single pin to the coupling capacitor (which  I changed for another 100nF in the larger 0805 size), on the side of the LNA. It's ugly, it's difficult, but it works with little apparent degradation of the performance.

virendera, 2019-01-14:
[image]

In this image can you show me the connection of SMP1330  so that at later i can protect hackrf.

cjheath, 2019-01-14:
Roughly: 
[image]

Any retro-fit like this will affect high-frequency performance - more so the longer your connections are. Even the PCB change in my PR will misbehave from 1GHz up - I need to re-route the signal paths and adjust trace widths to maintain continuous impedance.

virendera, 2019-01-15:
I think this improvement is only possible in updated PCB design. direct modification to  previous  older design will degrade performance.

virendera, 2019-02-22:
@cjheath can you tell me is there any other lna mmic chip which can be use as substitute.

virendera, 2019-02-22:
There are minicircuit mmic chip are they compatible, spf5043z lna chip too. These chip can be use for rx and tx lna.

virendera, 2019-02-22:
also tell me where to procure SMP1330. Is it possible to use bav99 in place of smp1330.

cjheath, 2019-02-22:
I found both the Skyworks diodes and the MGA MMICs cheaply through
AliExpress. The BAV99 would protect, but would retard performance more
(though I haven't checked the actual numbers).

virendera, 2019-02-23:
send me a link for smp1330

virendera, 2019-02-23:
it's my second lna which I replaced with soldering iron with lack of knowledge and first try my hand on smd component. It's a souvenir for me. It is one which damage when I checking antenna while it's running on and antenna touch with a rg6 wire  and megic smoke happen.

cjheath, 2019-02-23:
Correction: it was a different Skyworks part I got on Aliexpress. You can
get SMP1330 here:

https://octopart.com/smp1330-005lf-skyworks+solutions-3998647

virendera, 2019-02-25:
@cjheath are you know about impedance matching (50 ohm) and how it is achieve in kicad.

cjheath, 2019-02-25:
Yes but I am not your tutor and this is not the right forum. Please stop.

Do1SKL, 2019-03-04:
Hello

my LNA is broken, which is the right LNA
MGA 81563-Tr1g or MGA 81563 BLKG?

cjheath, 2019-03-04:
Looks like same thing from a different manufacturer. Definitely the same package so will fit. -BLKG is from Broadcom, the other from Avago. You can compare the data sheets here: https://octopart.com/search?q=MGA81563

ctakes, 2019-04-16:
cjheath,

I appreciate what you've written up here. You present a well thought out and sound argument for Mike to consider.

Would you by chance know of either a source for the MGA 81563-Tr1g LNA or a compatible equivalent that isn't discontinued? I'd certainly like to get my Hack RF up and running again, however, the apparent availability of this part is the limiting factor for me here.

Thank you in advance for your input.

cjheath, 2019-04-16:
@ctakes Thanks for your kind comments. I've only been doing RF design for ten years, but have the advantage of tutoring (and review of my [pull request](https://github.com/mossmann/hackrf/pull/572)) by a giant, who has RF design patents going back decades. Mike would do well to consider his input, but unfortunately he still doesn't admit the problem and has not even acknowledged the PR.

I bought some replacements MMICs
[here](https://www.aliexpress.com/item/10pcs-lot-MGA-81563-TR1G-MGA-81563-MGA81563-MARKING-81-SOT-363/32918314525.html)

You can of course just leave the LNA off and use the bias tee to power an external one - assuming your Tx amp isn't also damaged?

tomevans80, 2019-08-27:
Going right back to the top of this thread, is there a fairly straightforward (e.g. test points without desoldering anything) way I can test if U13 is also blown on my device. I have similar symptoms.

cjheath, 2019-11-27:
@tomevans80 They tend to fail short. If your multimeter can see a DC path between the input and output of the LNA (connect output side first and earth yourself and HackRF beforehand) then it's probably blown. But basically if you don't see signal and noise floor rising +14dB when you enable the LNA (perhaps instead dropping a little) thats a pretty clear indication

Allasso, 2020-03-18:
In March 2020 what is the status of this?  Was the HackRF One ever upgraded with better antenna port overvoltage protection?  Or is -5 dBm still the max?

evanbarn, 2020-07-09:
Just a comment. I've been using the HackRF One for 4 weeks now. Now, for the second time, both amplifiers are broken. I'm not sure what caused this, but I did probably touch the antenna. Will repair this again. I bought 10 spare amps for this. A protection circuit is very much needed!

cjheath, 2020-07-09:
I am currently finalising a revision of the design with robust protection included. Apart from a better protection system that I proposed previously, it also has a number of small improvements (e.g. better Bias T), part updates, and changes for ease of manufacturability (only 6 drill sizes not 13). All the circuit changes are under the existing RF shield can, and no external connectors have been moved, so things like PortaPack will still work ok.

evanbarn, 2020-07-13:
I'm very interested in the final protection.

cjheath, 2020-07-14:
The finished schematic is here. There are plenty more things that could be improved (e.g. lots of digital noise injection still), but I think I've done enough for this round. https://github.com/cjheath/hackrf/blob/add_protection/doc/hardware/hackrf-one-schematic.pdf

Please review the schematic and the commit history and provide comments.

cjheath, 2020-07-26:
@mossmann I have completely re-implemented my previously-proposed changes, and merged back to my master branch. The changes include a re-implementation of all your h1r5 changes (which I found after re-doing them - and had chosen the same MMIC, etc!) except the bias T power switch. I particular, I have improved the bias T itself, grounded the USB cable shield, and reduced the number of drill sizes to just six (it was 14). You will see the protection in my schematic, using clamp diodes that are designed and specified as limiter diodes for this purpose. I hope you will consider adopting these changes. Sample boards are being produced now, and assuming they pan out, will be available for purchase in a month or so.

cjheath, 2020-08-02:
Apparently I failed to merge back to master branch. The current version is still on my add_protection branch.

sh123, 2020-10-14:
I also damaged preamp and had to replace it, it happens to many people. Incidentally, transmitted from handheld, while forgot that hackrf was powered on. There are rumors in the Internet, that some hackrf boards have bad sensitivity compared to rtl-sdr, probably these people just fried their preamps, but not aware of it.

Localacct21, 2021-10-14:
What's the status for this? Why Is the pull request being held open in this matter @mossmann @straithe @cjheath

Localacct21, 2021-10-14:
@straithe thank you for at least giving us some point in reference. I appreciate the fact you took time to respond. We just wanted to hear from @mossmann himself. We do appreciate what you are doing for GSG. I have personally seen your work on Twitter and other social media forums.

straithe, 2021-10-14:
Thank you. One of the reasons Great Scott Gadgets hired me was specifically so I could be dedicated to responding to issues. This frees up @mossmann so he can have more time to do things like write code, project manage all of our hardware and software, and review PRs (just like you all are asking for). This is starting to get a bit off topic though, so I will be happy to talk more about the work I (and others) are doing for GSG on Discord.

teixeluis, 2021-10-20:
By all means, this is an issue that should be looked into. My perception is that HackRF is popular today for its fragile RF frontend. I guess as a design improvement this should be on the top of the list, and highly beneficial for the value of the product.

mossmann, 2021-10-21:
I've opened #974 to try to characterize the susceptibility of the RF amplifier to damage. This issue proposed a solution without characterizing the problem and without any evidence that the solution solves the problem and does not degrade performance. Some of the concepts in this discussion have merit and may be revisited after we have a clearer understanding of how the amplifier can be damaged.

closed 2021-10-21 by mossmann