greatscottgadgets/hackrf #543: hackrf constantly appearing and disappearing from USB?

nardev, 2018-11-03:
I understand that it's less likely a hackrf issue, but just in case if somebody had this issue before.

When plugged in, hackrf loops between being detectable and disappearing.

    $ hackrf_info 
    Found HackRF board 0:
    hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)

i'm using debian, kernel 4+

Any idea?

nardev, 2018-11-06:
I'll, most likely.. although it was working on this very port before and actually i already tried other  ports.

cjheath, 2018-11-06:
Grab the output of _dmesg_ and see what it says about the USB enumeration.

dominicgs, 2018-11-30:
@nardev Any update on this?  Did you get the output from `dmesg`?

nardev, 2018-11-30:
@dominicgs unfortunately no, will try do at Sunday.

luoGanda, 2019-11-28:
in my case, this was a bad usb connector - some are not as tight as they should be.
So i just put into another usb and it worked.

closed 2019-11-28 by miek