greatscottgadgets/hackrf #544: USB(?) interference via antenna from the lowest frequencies to above 190 MHz

lastfuture, 2018-11-05:
### Steps to reproduce
1. Connect Ant500 and extend it
2. Connect HackRF One with stock USB cable that came with HackRF One
3. Tune GQRX or GNU Radio to 90-100 MHz
4. Set gain to 24 dB IF and 30 dB BB
5. Observe FFT or waterfall

### Expected behaviour
smooth noise floor, random irregular noise pattern if any

### Actual behaviour
regular spikes and spurs and static noise characteristics as can be seen here
[image]

Also very distinct noise around 480 MHz where USB 2 has its clock frequency
[image]

### Additional info
The strength of both the regular noise patterns and the peaks around 480 MHz can be influenced by covering the cable with my hand, where covering the cable makes the noise pattern more prominent but reduces the intensity of the peaks around 480 MHz.

The noise is not visible when no antenna is connected, but it will always roughly scale with the strength of FM reception, even with antennae that are not tuned to the FM spectrum.

The noise will also entirely disappear when connecting HackRF to analog cable directly, even though the FM stations on there have a similar signal strength to the ones picked up via antenna. The noise floor is then smooth as expected

The computers I tested with are:
* iMac late 2012
* MacBook Pro 2018 via USB Type C plug to USB Type A socket adapter, both on battery and wall power

Noise intensity and characteristics are very similar on both machines

### Version information
**Operating system**:  
MacOS 10.14

**hackrf_info output:**  
```
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2b53b45f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006e475a
```

Gqrx version 2.11.5  
GNU Radio 3.7.10.1

lastfuture, 2018-11-05:
I've purchased a 20cm shielded USB Type C to Micro USB cable in an attempt to reduce both length of the cable and amount of plugs in the hopes it would make the phenomenon better.

Yikes! It makes the problem about 20 dB worse than the USB Type C to USB Type A adapter + USB cable included with HackRF.

Any suggestions as to how I can remedy the interference problem would be greatly appreciated.

unixpunk, 2018-11-05:
Have you tried this with another antenna yet?  Or an antenna (connected via cable) that is not near the device/computer like the ANT500 is by its nature?

lastfuture, 2018-11-06:
Yes. I've tried an antenna stand with 2m RG58 cable that came with my NooElec NeSDR Smart kit, similar to the one pictured here:
[image]

I made a conscious effort to stretch the cable as far as possible with the computer and the HackRF across the room

* with their telescopic antenna extended by the same amount
  * the result was the same as with the ANT500 connected directly
* with their differently tuned antennae
  * the FM signals were weaker and the spurs were weaker by the same amount, resulting in roughly the same dB difference between signal and noise

I've also tried walking around with this solution in hand using the telescopic antenna. Curiously the noise always seemed to increase/decrease with the FM signal strength.

I can very easily and repeatedly demonstrate that the noise gets considerably worse as soon as I put my hand on the USB cable. This screenshot shows me repeatedly touching and letting go of the USB cable. The same doesn't happen when I touch the coax cable
[image]

lastfuture, 2018-11-06:
Another test: This is me touching the USB cable first diectly at the computer, then increasingly closer to the HackRF in steps of about a hand's width apart. You can see the amplification of the interference gets worse the closer to the HackRF I touch the USB cable. None of this can be seen when I touch or cover the HackRF directly with my hand, or when I touch the coax cable.

[image]

cjheath, 2018-11-06:
Try with the HackRF and laptop sitting on a metal bench (or kitchen sink?) to provide capacitive ground coupling between them. This will reduce pickup via the USB cable. Also, is your laptop running on batteries or power pack, and have you turned off other potential sources of local interference?

lastfuture, 2018-11-06:
Capacitive ground coupling via a kitchen sink improved the noise significantly! (Moving away from any possible RF interference did nothing, I tried in the middle of a cemetary)

[image]

Alright! What can I do to make this effect more permanent? I hope I don't have to carry a sheet of metal with me everywhere I want to use my HackRF.

lastfuture, 2018-11-06:
It makes no difference if the laptop runs on battery or is plugged into wall power by the way, forgot to answer that part

N8AUM, 2018-11-06:
> Dang, I'm sorry I have to drag this out, but the previous test was with the sink at work.
>
> The sink at home as well as a metal baking tray have no effect. If anything they make it worse

Buy one of those cheap usb dongles and see how it reacts.Good luck,N8AUM

lastfuture, 2018-11-06:
I have an RTL-SDR dongle. It exhibits the same kind of interference.

Something else I just tried: I got my fingers on an FM band stop filter, and it eliminates the FM signals as expected, and also the interference along with them. That's both with the ANT500 and with the telescopic antenna connected via 2 meters of coax cable.

I've also tried on a third computer now, a 2016 MacBook. The interference looks the same there as well. The interference also looked almost the same at work on my MacBook Pro, several km from here. I'm not the only one using Macs with HackRF though, so this can't be a Mac thing, right?

lastfuture, 2018-11-06:
Just out of curiosity, how does the FM band look for you people around 97 MHz with the amp off, IF gain on 24 dB and BB gain on 30 dB?

unixpunk, 2018-11-06:
I checked this last night and for me the band looks similar to your screenshot for when you put things on the sink at work.  Quite except for the actual stations, no spikes every N khz or whatever.  I'm running PC though and my HackRF is in a metal case that houses the portapack with it.  I also don't have the original cable, I use a cable from monoprice that has a built in ferrite choke and gold-plated connectors.

lastfuture, 2018-11-06:
My HackRF is in the plastic enclosure it's usually sold with. Do you think things could improve with it in an extruded aluminum enclosure?

unixpunk, 2018-11-06:
Do you have the shielding/cage added to the hackrf board?  I would suggest doing one of the two, either add the cage/shield or get the metal case.  One option is more work but far less money than the other.  (I have both)

lastfuture, 2018-11-06:
I'm prone to damaging electronics when I do fiddly work on them, so I'd rather shell out extra money for the enclosure and not worry about any damage I could deal. Installing the shield sounds hard

cjheath, 2018-11-06:
If you want to know whether a shield will help, try the HackRF inside a cake tin (cut small holes for the USB and antenna leads). But this is really only for externally radiated noise, probably not your issue. The on-board shield protects the receiver from noise radiated from the rest of the HackRF.

However, your symptoms are caused by a noisy PC/laptop (noise on the USB ground), and lack of an earth on the HackRF. Isolate the HackRF with a CMC and earth it, use a CMC on the antenna too if you can, and you'll find the problem vanishes.

lastfuture, 2018-11-06:
I tested if I could find surfaces in the house and in the street where reception would get better like on the sink at work, but all I got was various patterns of interference

I'm posting these in the hope that someone recognizes a pattern that suggests a solution

HackRF and Laptop on the hood of our car (signal went crazy)
[image]

In the middle of the street holding the HackRF an arms length away from the laptop, then close
[image]

On a solid wood surface
[image]

Putting it and the laptop onto the metal railing of our balcony (signal went semi crazy)
[image]

On various surfaces in the kitchen (sink, stovetop, press board work surface)
[image]

cjheath, 2018-11-06:
You can stop with the location tests! I suggested the sink because it's probably earthed (your home must have plastic plumbing) to see if it looked like conducted or radiated noise. You clearly need an earth, and to isolate your HackRF from the noisy laptop.

lastfuture, 2018-11-06:
Thank you @cjheath 
As english isn't my first language, with station earth you mean earth as in ground from e.g. plumbing, right? Where would I connect it to? The SMA thread?

lastfuture, 2018-11-06:
If I understand the schematic all three SMA connector threads share common ground with the rest of the device, so I could make a grounded SMA connector minus the center pin and connect it e.g. to clkout, correct?

cjheath, 2018-11-06:
Plumbing is a good start, otherwise your house power earth (connect a single short earth wire into a 3-pin socket). The best place to connect on the HackRF is the shield of the antenna input. You could solder an earth connector to the base of the antenna SMA socket.

First though, I'd be looking for a USB cable with a CMC built-in (the lump in the cable). Does yours have this?

lastfuture, 2018-11-06:
It doesn't, the device came with a cable that has no lumps. I do have one ferrite core but if I understand correctly the choke has wire wound around it, which obviously is a difference.

I'll be looking for a cable with chokes on both sides as you described

cjheath, 2018-11-06:
Passing the USB cable even once through a toroid will help, better if you can do two or three turns. Or use one like these:
[image]

dominicgs, 2018-11-07:
I agree with @cjheath, I strongly recommend using one of those cables if you see this issue.

lastfuture, 2018-11-07:
I found a Micro USB cable with two chokes on either end today and was excited, but it makes the problem worse than it is with the included cable.

included cable:
[image]

cable with chokes:
[image]

These screenshots were taken seconds apart in the same location with the same settings.

I also tried a makeshift ground wire, holding one end to the ground of an outlet and the other to the thread of the antenna SMA connector, but either I wasn't able to make a connection or I did make a connection but it made no difference.

Can I be doing anything else wrong here? I feel like this has the potential to be really stupid user error given that none of the advice worked so far and you're very likely operating HackRFs much longer than me.
Just to make extra sure I'm not messing something up on the software level here, these are my settings. Can somebody please confirm that my Gqrx setup looks sound?

[image]
[image]
[image]

(more Msps make it worse, fewer make it slightly better)

lastfuture, 2018-11-08:
I've tried six more cables today, with choke, without choke, direct USB-C to Micro B, via adapter, via powered and unpowered hub. This is the absolute cleanest signal I can get and I'm super frustrated

[image]

I've also tried a fourth laptop with not much more luck

James-Brown, 2018-11-08:
Peter. I haven’t followed this whole thread but does the ‘hair’ go away when you turn off the RF Amp? If so I would bet of an amplifier that is oscillating.

From: Peter Marquardt 
Sent: Thursday, November 8, 2018 10:50
To: mossmann/hackrf 
Cc: Subscribed 
Subject: Re: [mossmann/hackrf] USB(?) interference via antenna from the lowest frequencies to above 190 MHz (#544)

I've tried six more cables today, with choke, without choke, direct USB-C to Micro B, via adapter, via powered and unpowered hub. This is the absolute cleanest signal I can get and I'm super frustrated

I've also tried a fourth laptop with not much more luck

lastfuture, 2018-11-08:
All of this is with the RF amp off. If I turn the RF amp on it looks exactly the same but about 10-15 dB stronger

barto-, 2018-11-08:
You could try to install an optional RF shield. Check this video https://youtu.be/IQ_I8q3IUFs

lastfuture, 2018-11-08:
To summarize:

The cleanest signal is still with the included USB cable. All other USB cables I've tried now, including the one with two chokes, produce a noisier signal. There's only one other cable in the bunch that also performs the same as the included cable.

A powered USB hub didn't help.

I've not been able to get a clean signal without the spurs on five computers, four of them laptops, four of them running macOS and one running Ubuntu.

Gqrx and GNU Radio agree on the FFT spectrum and they both show the spurs. I can also clearly hear it in the FM demodulation of weaker stations

The spurs disappear with no antenna connected or with a cable connected directly to the analog TV cable that also serves FM radio stations. Those come in super clear with no spurs in sight and a baby smooth noise floor.

The spurs are always there when an antenna is connected (except when a FM band stop filter is present), even in different locations (e.g. at home, at work) and with most electronics around turned off or far away from other electronics except the laptop. There is also no difference whether I use the ANT500 connected directly or another telescopic antenna connected via 2m long coax cable.

It happens with the laptop on battery power and powered via wall plug.

An RTL-SDR dongle exhibits similar (but differently spaced) spurs.

… taking all of this into account there are a few common denominators left that I can think of.
* I've performed all the tests in the same city.
* I'm always nearby when testing.
* There's always an antenna connected. (except for my test with cable)
* It's always a USB connection (and I think always capable of USB 3.0 or faster).
* I've performed all tests between 6 PM and 3 AM.
* I'm almost always using Gqrx and the same HackRF One. (except for my tests with GNU Radio and the test with the RTL-SDR dongle)
* The spurs always start tapering off above about 190 MHz until they're mostly gone at 210 MHz
* The HackRF has no RF shield and is in the injection molded plastic case

lastfuture, 2018-11-08:
Here's one more thing I've just tested and which got me thinking:

The difference between the following two screenshots is that for the first screenshot I've misused an AM loop antenna (at 24 gain IF, 30 gain BB) and in the second one I tried matching the energy and FFT as closely as possible with the ANT500 extended (at 16 gain IF, 18 dB gain BB)

[image]
[image]

The spurs are almost absent using the antenna meant for AM radio, even at higher amplification levels.
This got me thinking. I've read that an antenna can pick up signals well if it's the length of one wavelength, half a wavelength or a quarter wavelength. (Entering solid half-knowledge territory here)

* The AM antenna is about 3.5m long, which must roughly make it good at picking up 85.6 MHz, 42.8 MHz and 21.4 MHz
* ANT500 is about 72 cm extended. That means it would be roughly tuned to 416 MHz, 208 MHz and 104 MHz

Could this mean the spurious signals interfering with my HackRF must come from around 208 or 416 MHz since they're absent with the AM antenna?

Continuing that thought this would strengthen the theory that the component actually picking up the noise is the antenna, and in the light of everything else I've tested I know it's highly dependent on the cable, suggesting the cable is transmitting the noise. Since the USB 2.0 bus frequency of 480 MHz still seems to be the likely source and regarding the cable as a transmitting antenna, and assuming similar wavelengths calculations from above apply to sending antennae it would make sense that the length of the cable has something to do with the amount of noise present (along with its shielding I assume, but I can't measure that easily).

* The cables with which I got the best signal to noise ratio are 80cm and 95cm. Would that make them not good for transmitting 480 MHz?
* The worst cables were short, 25cm and 20cm. Could they be much better at transmitting close to 480 MHz?

Does that help me solve my problem?

lastfuture, 2018-11-08:
… and following that train of thought, which length of USB cable would be the worst possible at transmitting 480 MHz?

cjheath, 2018-11-09:
The spurs are probably all harmonics. Of whatever the spacing is. What is
it?

lastfuture, 2018-11-09:
It looks like they're pretty much spaced 48 kHz apart, which incidentally is also the audio output's sample rate. What does that mean?

lastfuture, 2018-11-09:
I have to revise my answer, those are the coarse spurs, if I increase the HackRF's sample rate I can also see fine spurs which fall into an 8 kHz raster

lastfuture, 2018-11-09:
Here's a close up of the spurs. The ones at 48 kHz intervals are fanned out while the ones at 8 kHz intervals are sharp
[image]

lastfuture, 2018-11-10:
I've had the unexpected opportunity of talking to the former Director of National System Development for Cellular One about this. He said this is very likely heterodyning. He said my working theory about the USB cable transmitting (resonant close to a quarter wave of 480 MHz) and the ANT500 receiving (close to a half-wave of 480 MHz) isn't far off. I'm on the right track about varying the length of USB cable in order to reduce the interference but since we don't know the primary heterodyne component I'm left to trial and error.

So I've decided to introduce as much trial into trial and error as I can. Perhaps I can build a variable length USB cable.

I'm learning a lot about RF and It's exciting

cjheath, 2018-11-12:
I have the RF shield off mine because I expect to replace an MMIC soon. I see a lot of regularly-spaced (8KHz) spurs filling the frequencies between 27MHz and 33MHz. I think they're very likely from an internal source. Also spurs on 10MHz multiples running up to a couple hundred MHz (possibly from the new TCXO I just added?) I'll let you know what I find as I experiment further. I haven't tried connecting one of my large external antennas yet, for example.

lastfuture, 2018-11-12:
I see the 10 MHz spurs you describe as well.

The strength of my spurs in the FM band seems to be very dependant on the antenna connected, and they disappear in certain cases. I'm therefore assuming that it's not an internal component causing them

lastfuture, 2018-11-12:
I've tried two 1.8m USB cables that resulted in terrible noise. The USB cables yielding the best signal to noise ratio so far are 1m and 90cm. I'm going to try to confirm my cable length theory with more cables close to 1m in length.

I'm also braving up and installing an RF shield, even though my observations point towards it coming through the antenna. My soldering skills are not the best but I figured it should be doable and an RF shield certainly won't make it worse, right?

cjheath, 2018-11-12:
You're probably aware of this, but a more efficient antenna that's *not* receiving the spurs will push them down in relative level. So you need to look at the absolute level to know what's going on.
Also, making a shield doesn't require soldering, if your board has the clips for it. It just requires some accurate metal bending and careful trimming.

lastfuture, 2018-11-12:
So far three telescopic antennae which were all tuned to somewhere in the FM spectrum received the spurs at pretty much the same absolute level. My point of reference is one particular FM radio station that just about comes through the spurs with the best USB cable. I'm judging my USB cables by both the absolute intensity of the spurs in dBFS and by the intensity of my reference radio station in relation to the peaks of the spurs (and audible quality of the WFM demodulated sound)

My observation so far is:
* My USB cables close to 1m yield spurs peaking at around -80 dBFS and the radio station is visible in the noise and has some static but can be heard OK
* The USB cables much longer or shorter than 1m yield spurs peaking at around -65 dBFS and the radio station is harder to see in the FFT as well as much more staticky all the way to undetectable

To be honest I don't know if my board has clips for it or how they would look. With clips do you mean the holes in between the solder pads as can be seen in this image? https://camo.githubusercontent.com/fc5fea2c0ccc11b91ce651b812bc1d6cd0fbbfc9/68747470733a2f2f7261772e6769746875622e636f6d2f6d6f73736d616e6e2f6861636b72662f6d61737465722f646f632f4861636b52462d4f6e652d6664302d303030392e6a706567

cjheath, 2018-11-12:
I assume you aren't using AGC (input settings) but setting RF and IF gain manually.

I mean the clips on the dashed RF boundary trace in this photo.
[image]

lastfuture, 2018-11-12:
Yes, always the same gain settings: no RF gain, 24 dB IF and 30 dB BB. Those are the gain settings that make my reference FM radio station come in the clearest with the good cable. I'm also always cross-referencing against the good cable just in case other outside effects (e.g. atmospheric phenomena) changed how well this particular station is reaching me.

I didn't know some boards had these clips. I'm going to open my HackRF's case when I get the shield so I won't know until probably Saturday, unless there's another way to find out

lastfuture, 2018-11-14:
OK so this is what my FM radio spectrum noise floor is supposed to look like, huh?

My sample size of possible USB cables has grown so much that I can now confidently say 0.9-1.0m length seems to be close to the sweet spot, 1.6-1.8m are terrible and 0.2m-0.4m are also terrible, all pretty much regardless of cable quality.

I'm pretty certain I'm not the only one using a HackRF and its included USB cable with a recent 2018 MacBook Pro Touch Bar, a 2012 iMac or a 2016 MacBook. Shouldn't there be more reports about this in that case?
Since those computers all show the spurs all the time in repeated tests … could my HackRF be defective?

lastfuture, 2018-11-14:
Speaking of TCXOs … my HackRF is off by -25ppm, could that be related to the spurs?
As far as I can see a new TCXO is plug-and-play. Put it on the respective pins and it's in use instead of the built in oscillator. It couldn't hurt to try one, right?

cjheath, 2018-11-14:
Right. Not quite plug and play, I had to bend two of the ISP pins down to
clear the oscillator can. I don't know if TCXOs are available with
different geometry. If I need to use ISP (hopefully not) I'll fit
right-angle header pins out the back of the board.

cjheath, 2018-11-14:
Your problem is not your cables, they're just changing how you see the problem. Neither is lack of a TXCO I reckon, or you'd see a more variable result (drifting spurs, varying spacing, etc). It's most likely something about your computer or your location. Take stuff to another suburb, and try a different computer. If the problem goes away, your HackRF is probably fine, otherwise it's busted. It could be a simple matter of bad power supply bypassing (dud cap, or bad soldering). You might be able to fix it by reflowing the board...

mossmann, 2018-11-16:
cjheath wrote:

> I mean the clips on the dashed RF boundary trace in this photo.

I've never seen clips like that before.  I didn't design for them, and we don't install them on Great Scott Gadgets boards.  I don't know who made your HackRF One, but its origin may have something to do with why it doesn't perform at lower frequencies.

I designed the RF section of HackRF One to fit the Laird BMI-S-230-C RF shield on the BMI-S-230-F-R frame.  We do not install the shield or frame on the units we sell.

mossmann, 2018-11-16:
lastfuture: Everything you have reported appears normal to me.  It is common to pick up USB interference with a USB-connected radio receiver such as HackRF One, and most of that interference should be between 0 and 240 MHz.  It is common for changes in USB cables, antennas, grounding, and coupling with nearby objects to affect how this interference is conducted/radiated/received.

I do not believe your HackRF One is defective.  (If you were seeing interference above -40 dbFS with moderate gain settings I might think differently.)  It is unsurprising to me that you see a similar interference pattern with different spacing when using an rtl-sdr dongle.  HackRF One has a higher  maximum USB data rate, so it should typically result in more spurs per bandwidth.

The USB cables we include with HackRF One were selected based on some testing I did which indicated that the quality of shielding was much more important than the quality or presence of a ferrite.  Be aware that some USB cables have better shielding than others, and many USB cables (especially those intended primarily for charging) have no shielding at all.  You can test the shielding with an ohmmeter.  Also be aware that some USB cables that appear to have a ferrite actually just have an empty shell where a ferrite could go.

lastfuture, 2018-11-16:
@mossmann thank you, your assessment of my situation means a lot. I guess I had too high expectations after seeing much cleaner spur free noise floors in videos showing off HackRF.

The spurs definitely feel like they're sullying signals I could otherwise demodulate. e.g. I've not managed to even find more than one AM radio station, and that station was barely distinguishable from the spurs and came in just clear enough to barely make out its name when it was announced. I've not been able to find the signal in the noise again, even though I've bookmarked its frequency, and even though I'm now trying with an antenna that should theoretically be tuned to roughly a quarter wavelength

cjheath, 2018-11-17:
@mossmann Thanks. Sad that the HackRF I bought (2nd hand and damaged) might not be genuine. Nothing about it looks sub-standard, but I suppose that goes with the territory. I guess the mixer baluns are a spot where a clone-maker might skimp? Or perhaps the software is just not enabling the correct filter path. Anything else I can do to check what's causing low-band deafness? Or perhaps I just don't have an adequate antenna?

The PCB is labelled "13 February 2014".
`hackrf_info` says my serial number is 0000000000000000167c60c837524be7
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa0000a30 0x00624f5d, does that look plausible?

@lastfuture perhaps try to use a better antenna? The best first answer to SNR problems is "get more signal" :)

lastfuture, 2018-11-17:
OK, I've installed the full monty now. TCXO, RF shield, aluminum enclosure. A first test with the ANT500 is looking very promising. A LOT better than in its factory state in the original case.

The TCXO is bumping the HackRF from -25ppm to 0ppm, and I managed not to fry it while soldering the RF shield, so hooray.

[image]

This time for the first time the spurs aren't stable and changing intensity. I'm not entirely sure what's going on or what could be causing it. The screenshot above shows the absolute best that the s/n ratio gets. The worst looks almost like the best before the modifications. It seems to be somewhere in between for the most time.
I'm going to leave this running for a few hours away from the laptop with a slow waterfall and take a look at it later to see if it's changing without me being around.

lastfuture, 2018-11-17:
There was no difference after a few minutes, the spectrum stayed in the "somewhere between" I mentioned.

However I learned that the difference between these two orientations of the HackRF …
[image]

… results in this difference in spectrum.
[image]

I'll have to be extra careful how I position it

mossmann, 2018-11-19:
cjheath: It may be a genuine GSG board that has been modified. If it is one of our boards it should have the GSG gear logo visible on the bottom copper layer.  If it is not a GSG board, the most likely component substitutions that would affect the low frequency performance would be the DC blocking capacitors and inductors in the RF section.

Use of a proper antenna for the frequency of interest is important for performance.  ANT500 is not appropriate for anything below about 75 MHz.

It is expected for the performance at 1 MHz to be about 10 dB worse than the performance above 10 MHz.  See: https://greatscottgadgets.com/2015/05-15-hackrf-one-1MHz/

mossmann, 2018-11-19:
lastfuture: Congrats on the successful mod! The results from that change of orientation are fascinating.

lastfuture, 2018-11-19:
I'm still seeing a lot of spurs all across the spectrum (with the ANT500 and a similar telescopic antenna in particular), some of them making it harder for me as a learner to separate proper signals from the noise. e.g. this is my spectrum around 433 MHz at much more reasonable gain settings of RF off, IF 16 dB, BB 16 dB

I know that the ANT500 is probably not the best antenna for that frequency range, but can you please take a look at these and tell me if they're supposed to look that way?

at 20 Msps
[image]

at 12.5 Msps
[image]

at 8 Msps
[image]

I'm assuming all of the spikes I see are spurs as they're permanent and don't _seem_ to be signals. Also quite a few of them are moving into the opposite direction of the direction i'm tuning, and they're mirrored around the center frequency, making me believe they're some sort of aliasing artifacts (?) The actual signals in this ISM band (mostly outside thermometers of neighbors' weather stations) are coming in at around 20 dB quieter than the loudest spurs.

If I'm honest I'm a bit disappointed with my HackRF experience right now. I had hoped for a cleaner spectrum for me to learn in. I'd still really love to do more to get it cleaner and spur-free. It's also making it hard for me to be a good neighbor on the spectrum as I learned I should be, because I need pretty CPU intensive settings on my band pass filter in order to prepare signals for replaying while leaving out the spurs. So steep in fact that I have to prepare them offline

mossmann, 2018-11-20:
@lastfuture: I really don't think you have a problem with spurs.  If you go looking deep into the noise floor with any radio receiver, you will find such things.  I think you may have a problem with the received signal strength of your signals of interest, however.  Signals such as your neighbor's weather station should be detectable at a higher power than the spurs.

Do you happen to have an RF signal generator or a friend with another HackRF that you could use to perform a measurement?

lastfuture, 2018-11-20:
Unfortunately I have neither an RF signal generator nor a friend with a HackRF. I do have a few remotes that will generate signals but I have no way of knowing in which intensity. I could compare the same remote received with the HackRF and with a NooElec neSDR SMArt if that would help

unixpunk, 2018-11-20:
@lastfuture You mentioned the TXCO, that could be the noise you see every 10mhz.  Unless you're doing specific things where having a perfect click is required, might try taking that out of the mix.  Will you have to adjust ppm offset then?  Sure, but once you know what that offset is when the device is up to temp (by comparing to a known-stable signal), it usually doesn't change much.  As Mossman mentioned and as your LTE band antenna points to further, you may just need a better tuned antenna so that your desired signal is stronger and also so that other noise is tuned out.  The gains are also important, you may find that having one too high over the other causes some spurs to show or disappear even.  Sometimes having the LNA/amp on is better, then you can lower the other gains or vice-versa.  Hope that helps.

mossmann, 2018-11-20:
I agree with @unixpunk that the TXCO could be a source of some spurs.

The SNR of the DECT screenshot looks good at a glance, but I notice that none of those peaks is higher than the DC spike.  They should be if the DECT devices are at close range.

[5 more replies]

closed 2019-12-09 by miek