greatscottgadgets/hackrf #548: Poor signal power in GNU Radio

mpozzo, 2018-11-13:
Hi, I've tried a FM example with GNU Radio and I get a really poor signal strength. It's not a hardware problem, as I've tried with three different HackRF's and antennas.
When I use SDR# in Windows, the signal improves a lot, even using the same gains (IF gain 16 dB and BB gain 20 dB with amp off). Here's captures of both programs.

[image]
[image]

As you can see, there's a station around 99.7 MHz that gets -40 dB in GNU Radio and -20 dB in SDR#

I've tried with many versions (3.7.11, 3.7.9 and 3.7.13) of GNU Radio in Windows and Ubuntu, and also using the Live environment.

What could be the problem? Thanks!

Regards,
Maximiliano

kpreid, 2018-11-14:
It looks like you have approximately the same SNR in both cases (the height difference between the peaks and the noise floor is the same), so I would suspect this of being just a scale calibration issue. I'm not familiar with SDR# but a web search indicated that it has a configurable scale http://forums.qrz.com/index.php?threads/rtl-2832u-r820t-sdr-as-a-spectrum-analyser.391657/ that might well be different from GNU Radio defaults to. You can also change the reference level in the GNU Radio FFT widget for the same effect.

mossmann, 2018-11-16:
I agree with kpreid.  The reference levels on those plots are arbitrary.

closed 2018-11-16 by mossmann