greatscottgadgets/hackrf #549: LTE-Cell-Scanner

jjbaek35, 2018-11-17:
### Steps to reproduce
1. I tested open-source baed LTE-Cell-Scanner with HackRF one (https://github.com/JiaoXianjun/LTE-Cell-Scanner)
2. I installed HackRF host and LTE-Cell-Scanner successfully
3. When I run the  LTE-Cell-Scanner, it found device but not scan

$ CellSearch  -s 939e6 -e 940e6
LTE CellSearch (release) beginning. 1.0 to 1.1.0: OpenCL/TDD/HACKRF/bladeRF/ext-LNB added by Jiao Xianjun(putaoshu@gmail.com)
  PPM: 0
  correction: 1
HACKRF device FOUND!
Use  HW  begin with 939MHz actual 939MHz 1.92e+06MHz
    Search frequency: 939 to 940 MHz
with freq correction: 0 kHz
    Search PSS at fo: -60 to 55 kHz

Examining center frequency 939 MHz ... try 0

### Expected behaviour
It starts searching the frequency turn by turn.

### Actual behaviour
It doesn't pass to next frequency. 

### Version information
Ubuntu 16.04

**hackrf_info output:**
hackrf_info version: git-abc7d68
libhackrf version: git-abc7d68 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c8233c8a5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00724f5f

*If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output

mossmann, 2018-11-19:
This appears to be a problem with LTE-Cell-Scanner, not with HackRF.

closed 2018-11-19 by mossmann

jjbaek35, 2018-11-19:
I think so, Can I get old firmware made in at around 2013?

andreoua, 2019-01-28:
Is Ubuntu hosted on a VM?

Ririconj, 2021-03-10:
Is firmware downgrade helped?