greatscottgadgets/hackrf #550: Clocking divider in firmware

nvb13, 2018-11-19:
What firmware modification should i do to clocking hackrf by external 100mhz OTXO?
In what file i need change dividers?
Thanks!

nvb13, 2018-11-20:
Thanks! How I understand max. division is 8, so freq will bi 12.5Mhz.
I can use external divider by 4 so freq of ext generator will be 25Mhz, will hackrf work without changing firmware?

mossmann, 2018-11-20:
Ah, good point.  Instead of an external divider, you could use CLKIN_DIV to divide the input down to 25 MHz and then use the same multisynth settings as PLLA.

With unmodified firmware and a clock frequency other than 10 MHz on CLKIN, the sample clock and RF reference clocks will be incorrect, but the HackRF such function otherwise (e.g. you should be able to communicate with it over USB).

nvb13, 2018-11-22:
But i have already OTXO 100Mhz with divider x4 = 25Mhz. What i need to change here?
`/* 10 MHz input on CLKIN for PLLB */
	data[0] = 34;
	data[4] = 0x26;
	si5351c_write(drv, data, sizeof(data));`
Thank you! Sorry for my n00b question)

mossmann, 2018-11-23:
If you are feeding 25 MHz into CLKIN, just comment out this line:
`data[4] = 0x26;`