greatscottgadgets/hackrf #554: VBUSCTRL pin and USB connector voltage leakage

sgoudelis, 2018-11-22:
I am referencing issue #539 since it was closed. I have fed 5v into VBUSCTRL and there is still voltage in the USB connector. Have I understood correctly that feeding 5v into VBUSCTRL will stop the voltage leaking over to the USB connector ?

mossmann, 2018-11-23:
Sorry.  I misread my own schematic!  The MOSFET controlled by VBUSCTRL switches off current draw from the USB connector into the HackRF One.  It does not prevent current from flowing from the HackRF One out the USB connector.

sgoudelis, 2018-11-23:
Could I replace that MOSFET with something that would perform that function ?

mossmann, 2018-11-23:
I don't know of any part that would fit the same footprint that would do that.  You could install a diode in place of FB1 that would permit only inbound current from the USB connector.

sgoudelis, 2018-11-24:
Couple of questions: What type of component is FB1 ? I see in the schematic in KiCAD. So what you are saying is I could replace that with a diode and stop current going back into the USB connector and into the USB host. What kind of diode would you use ?

mossmann, 2018-11-27:
FB1 is a ferrite bead for reduction of noise that enters via the USB power supply.  Replacement of the ferrite with a diode may introduce a bit of extra noise while a USB host is attached, but it probably won't hurt much because there is still another ferrite in the circuit (on the GND signal).

I would use a standard rectifier or schottky diode such as  SD0805S020S1R0.

closed 2019-12-09 by miek