greatscottgadgets/hackrf #555: HackRF One not receiving

zs2hx, 2018-11-28:
Unit does not receive - dead out of the box - only noise.

Tested two other HackRF units and they are fine - so not a software issue.

hackrf_info version: git-953df27
libhackrf version: git-953df27 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8388f76cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005f4757

dominicgs, 2018-11-30:
Are you using an identical setup for all three?  Same antenna, USB cable, USB port, etc?

dominicgs, 2018-12-01:
Could you try reflashing the firmware, just to be sure?

zs2hx, 2018-12-03:
Tried that 3 times - no joy.

dominicgs, 2018-12-03:
Please contact the reseller that sold you your HackRF One and ask them to replace it.