greatscottgadgets/hackrf #556: HackRFOne not working with GQRX on SkyWave Linux

DRTAC7, 2018-11-29:
### Steps to reproduce
1. Click GQRX icon on desktop
2. Goto configure I/O device settings
3. Select HackRF One #(my devices serial number)
4. Press the start DSP button

### Expected behaviour
GQRX should receive data from HackRFOne, RX LED should light up on device. I should be able to tune to different frequencies using the GUI for GQRX

### Actual behaviour
Nothing happens. As if I typed xyzzy in the wrong place. ;)
Joking aside, actually nothing happens. I do not get a waterfall.
Under input controls, I do not see different options for amplification control.
Nothing happens.

### Version information
**Operating system**: Skywave Linux which uses the MATE Desktop Envrionment 1.16.2
https://skywavelinux.com/

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: xxxxxxxxxxxxxxx
Board ID Number: 2 (HacRFOne)
Firmware Version: 2018.01.1 (API:1.02)
Part ID NUmber: xxxxxxxxxx xxxxxxxxxx

GQRX Version 2.11.2

### Output
```
Insert any commandline or build output here
```

DRTAC7, 2018-11-29:
Update: Because of the complete (and understandable) lack of support for Skywave linux, I am considering switching to Ubuntu, as it seems that is the operating system that provides the most general support for SDR. I know that Mr. Mossmann suggests using Pentoo. Would anyone be willing to respond with their thoughts on using Ubuntu versus Pentoo and visa versa? Thank you.

dominicgs, 2018-11-30:
While Pentoo, like Skywave, comes with many tools preinstalled, it can be tricky for people not used to the Gentoo Linux world.  I recommend stock Ubuntu, it's what I run and it's what many HackRF users use.

However, it's unlikely to be the operating system at fault here, just configuration.

Can you run GQRX from a terminal and see if there is any error or information in the output?

DRTAC7, 2018-12-02:
Hi @dominicgs Thank you for responding!

Skywave was also giving me trouble with gnuradio-companion, so I decided to switch to Ubuntu 18.04 LTS. 

I installed GQRX from source on Ubuntu, ran it and encountered the same issues using the HackRF. 
Now, I am pretty sure I did not install GQRX correctly, or that an error occurred during the process. 

GQRX is also available via Software (what I think is the new name of synaptic package manager?)

I am still a novice when it comes to Linux. Would you be able to walk me through the steps you would take to installing GQRX correctly?

Thank you.

--7

dominicgs, 2018-12-02:
I would install GQRX from the Ubuntu software tool, that's the version that
I use.

pelixm, 2019-03-06:
Hi
I am facing exactly same problem under Ubuntu 18.04. Is it possible to reopen this issue? Otherwise please let me know how I can provide further details. Thanks.