greatscottgadgets/hackrf #559: some probleam with cpld code

hustpigeon, 2018-12-04:
[image]

here   process(host_clk_i)
    begin
        if rising_edge(host_clk_i) then    
            if codec_clk_i = '1' then
                -- I: non-inverted between MAX2837 and MAX5864
                data_to_host_o <= adc_data_i xor X"80";--高变，低不变
            else
                -- Q: inverted between MAX2837 and MAX5864---q=1 0x80,q=0 0x7f
                data_to_host_o <= adc_data_i xor rx_q_invert_mask;
               ////////according MAX5864, I think when Q invert ， here should be
            /////////data_to_host_o <= adc_data_i xor rx_q_invert_mask + **1;**
          ///////// addition 1 should be plused , for 127 and -127 are complementary
            end if;
        end if;
    end process;

jboone, 2018-12-04:
You're technically correct that the negation by inversion introduces a DC bias in the received samples:

Input | Raw | not[6:0] | Output
-------|-------|------------|----------
+127 | 1111_1111 | 1000_0000 | -128
+1 | 1000_0001 | 1111_1110 | -2
0 | 1000_0000 | 1111_1111 | -1
-1 | 0111_1111 | 0000_0000 | 0
-2 | 0111_1110 | 0000_0001 | +1
-128 | 0000_0000 | 0111_1111 | +127

However, if one were to add an offset to correct for it, you would then have to handle overflow for -128 -> +128. This is because +128 can't be represented by an 8-bit two's complement number. The obvious solution would be to saturate in this case, turning -128 into +127. That would require additional logic and delay through the CPLD. The CPLD is quite small, so it hasn't been practical to add saturation to the CPLD code. Besides, there's a significant DC offset from other parts of the radio. The DC offset added by the CPLD is minor in comparison.

If you still want to address this DC offset, I think the best solution is to remove CPLD negation and replace it with additional firmware on the LPC4320 M4 core. There's an interrupt which occurs when a block of baseband samples is available from the SGPIO peripheral. The code that moves that data could perform an inversion with saturation at the same time. The QSUB8 instruction should do the job.

jboone, 2018-12-04:
Oh, I forgot to mention, the even easier way to perform spectrum inversion (which is the ultimate goal of the CPLD invert feature) is to swap the I and Q samples. That might be possible in the CPLD, and certainly possible in the Cortex-M4 SGPIO interrupt. The REV16 instruction would do that, without the extra register and load instruction the QSUB8 solution would require. See this article for more: https://www.dsprelated.com/showarticle/51.php

hustpigeon, 2018-12-05:
> That might be possible in the CPLD, and certainly possible in the Cortex-M4 SGPIO interrupt. The REV16 instruction would do that, without the extra register and load instruction the QSUB8 solution would require. See this article for more: https://www.dsprelated.com/showarticle/51.php

> You're technically correct that the negation by inversion introduces a DC bias in the received samples:
> Input 	Raw 	not[6:0] 	Output
> +127 	1111_1111 	1000_0000 	-128
> +1 	1000_0001 	1111_1110 	-2
> 0 [...]
================================================================
Thank you very much @jboone , in one word, in thoery  plus one is right if it has more bits; in practice not plus one is also right for the error is just 1 ?

another question: why Q invert,  it has 2 reasion, 1: I Q wire reverse by PCB.  2: when mixed by mixer , if RF > LO   Q not invert , if RF<LO  Q invert, why not RF> LO Q invert and RF<LO Q not invert.  am i right ?

miek, 2019-11-25:
Yep, Q is inverted between the MAX2837 and MAX5864 so it is inverted again in the CPLD.

closed 2019-11-25 by miek