greatscottgadgets/hackrf #567: HackRf set-up: Trying to get the hackrf to operate in ubuntu linux terminal.

johnteoh90, 2018-12-24:
### Steps to reproduce

Installed the follow in ubuntu linux terminal
1. sudo apt-get install git build-essential cmake libusb-1.0-0-dev liblog4cpp5-dev libboost-dev libboost-system-dev libboost-thread-dev libboost-program-options-dev swig
2. git clone https://github.com/mossmann/hackrf.git
3. cd hackrf/host
4. mkdir build && cd build
5. cmake ../ -DINSTALL_UDEV_RULES=ON
6. make
7. sudo make install
8. sudo ldconfig
### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead
when run hackrf_info after the above steps, the following outcome appear. 
hackrf_info version: git-454e32b
libhackrf version: git-454e32b (0.5)
No HackRF boards found.

I have even tested by unplugging and replug the hackrf into my usb port.

3v3, 1v8, USB - green led light
RF - red led light
### Version information
**Operating system**:
win 10 64 bit
**hackrf_info output:**
hackrf_info version: git-454e32b
libhackrf version: git-454e32b (0.5)
No HackRF boards found.
If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

what I am trying to achieve.

I am working on school project that I hope to use the hackrf to capture a range of radio frequencies and export out the captured frequencies into a csv file. 

Hope someone can help me resolve this problem. Thank you.

mossmann, 2018-12-24:
Does it make a difference if you execute hackrf_info very quickly (within 1 second) after plugging it in and seeing the USB LED turn on?

Does it make a difference if you execute hackrf_info as root?

You mentioned both Ubuntu and Windows.  Which are you running?  Is there a virtual machine involved?

johnteoh90, 2018-12-25:
Does it make a difference if you execute hackrf_info very quickly (within 1 second) after plugging it in and seeing the USB LED turn on?

 **Tried this method and it is still returning this:
hackrf_info version: git-454e32b
libhackrf version: git-454e32b (0.5)
No HackRF boards found__**

Does it make a difference if you execute hackrf_info as root?

**no difference**

You mentioned both Ubuntu and Windows. Which are you running? Is there a virtual machine involved?
I am running a ubuntu terminal on a Windows 10 OS. No.

johnteoh90, 2018-12-25:
Hi @mossmann, sorry that I am quite confused with the installing guidelines provided in this wiki. 

want to check with you what would be the proper steps for setting up the hackrf on windows platform and using ubuntu terminal/window powershell to execute hackrf command lines. 

What I want to achieve is to execute this command "hackrf_sweep -f 2400:2490 > FrequencyScan.csv" 

Really hope you could assist me in this problem.

johnteoh90, 2018-12-26:
Hi @mossmann, I am running a linux subsystem on windows 10. Is there any work around it for windows OS platform to run hackrf-tools command line?

johnteoh90, 2018-12-26:
@mossmann I have been trying using mingw to compile the hackrf. Through this guideline provided in this link http://www.proxmark.org/forum/viewtopic.php?id=2126 by **asper contributor**  which is quite similar to your steps provided in this link https://github.com/mossmann/hackrf/wiki/Operating-System-Tips.

Is it possible to provide a clearer step by step guide for this case? Would really appreciate your kind help.

mossmann, 2018-12-27:
Have you tried installing the Windows executables for HackRF included in the PothosSDR installation?

johnteoh90, 2018-12-29:
Hi, @mossmann thanks for the suggestion and I managed to figure it out with your suggested solution. Cheers! Really appreciate your great help. 

However, sorry to trouble you again as I wish to understand the output that has generated from using the Hackrf sweep command. 

The following command  (hackrf_sweep -f 2400:2490 > FrequencyScan.csv) was used to generate the output shown in the picture below.
[image]

The question that I would like to ask is: 
1. when the hackrf perform a sweep it scanned a range of frequencies in a millisecond? That is why column C and D shows a lower and upper frequency of a range?

2. May I know what is the column E and F data?

3. Why is there 5 different values of signal strength column?

Hope you could help me clarify with this question which could really aid me in understanding the usage of hackrf better!

Thank you.

mossmann, 2019-01-03:
Yay! I'm glad the Windows binaries worked for you.

The output format is breifly explained at the end of the help:

```
$ hackrf_sweep -h
Usage:
        [-h] # this help
        [-d serial_number] # Serial number of desired HackRF
        [-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
        [-f freq_min:freq_max] # minimum and maximum frequencies in MHz
        [-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
        [-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
        [-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
        [-n num_samples] # Number of samples per frequency, 8192-4294967296
        [-w bin_width] # FFT bin width (frequency resolution) in Hz
        [-1] # one shot mode
        [-B] # binary output
        [-I] # binary inverse FFT output
        -r filename # output file

Output fields:
        date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .
```

1. Each range of 5 MHz (actually two ranges of 5 MHz) is analyzed all at once from the same set of samples, so a single timestamp applies to the whole range.

2. Column E tells you the width in Hz (1 MHz in your case) of each frequency bin (which you can set with -w).  Column F is the number of samples analyzed to produce that row of data.

3. Each of the remaining columns shows the power detected in each of several frequency bins.  In your case there are five bins, the first from 2400 to 2401 MHz, the second from 2401 to 2402 MHz, and so forth.

dominicgs, 2019-01-03:
I've added this as documentation on wiki: https://github.com/mossmann/hackrf/wiki/hackrf_sweep

closed 2019-01-03 by dominicgs

AdvaithSiva, 2020-05-03:
Hi @mossmann, I am trying to configure the hackrf one board on a Linux (ubuntu 18.02) virtual machine running on windows OS. I was wondering if it is possible to do so as I followed the readme file provided . My end goal is to use hackrf with GNU radio either on Windows or a Linux virtual machine. Could you please suggest what the easiest option is? 
I tried to follow the readme file provided here , https://github.com/mossmann/hackrf/tree/master/host 
This works well till I command hackrf-info which returns No hackRF boards found but the USB light on the board is green. Your help would be really appreciated.

Cheers
Advaith

AdvaithSiva, 2020-05-03:
Hi @mossmann , I got it working on the ubuntu virtual machine, turns out that i forgot to select the usb port. But it is still not working on windows. I used gnuradio to use the hackrf.

[image]

Attached is the screenshot of the error that shows up when I execute the blocks as shown in the video, https://greatscottgadgets.com/sdr/1/
It would be amazing if you would be able to help fix that or recommend a better way to connect hackrf one on windows which can used to communicate with a USRP B200mini-i SDR.

Thanks in Advance, @mossmann 
Cheers
Advaith

ChansimDeng, 2020-05-13:
> ### Steps to reproduce
> Installed the follow in ubuntu linux terminal
>
> 1. sudo apt-get install git build-essential cmake libusb-1.0-0-dev liblog4cpp5-dev libboost-dev libboost-system-dev [...]

problem：
hackrf_info:
hackrf_info version: git-43e6f99
libhackrf version: git-43e6f99 (0.5)
No HackRF boards found.

1.my env:
Physical machine：mac os
virtual machine：ubuntu 16(hackrf use)

and I also successfully installed the required libraries in ubuntu, eg:libusb-1.0-0-dev hackrf