greatscottgadgets/hackrf #568: OperaCake PCBA

matthiasben1, 2018-12-26:
Hello,

I'd like to make my own operacake board. However, I need some help.
 
My initial thought was to use the PCBA service provided by Seeed Studios. I've upload a BOM and the Gerbers, however I am not certain as to the PCB material selection.
 
1. You have recommended using FR408, but SeeedStudios let me choose between: FR4 TG130, FR4 TG170, and also something they call "High Frequency Board" which is able to carry over 1GHz signal. Which one is appropriate, if any?
 
2. You have specified 0.5oz internal copper (which conforms to the OSH Park specifications). Seeed will only let me choose between 1oz and higher (one value for all 4 layers?). Is this a problem in your opinion?
 
3. Alternatively, do you know any good PCBA manufacturer for small scale production ("maker" style), which will produce a _working_ operacake board?

mossmann, 2018-12-27:
We expect to be shipping fully assembled and tested Opera Cakes by March 2019, so perhaps that will meet your need.  If not. . .

The "High Frequency Board" is probably your best bet, but you would need to get stack-up details and use a microstrip impedance calculator to find out if you need to adjust the RF trace width to achieve 50 ohm impedance.

The two biggest problems you are likely to have with a prototype assembly service for Opera Cake are characteristic impedance of the RF traces on the PCB and lack of testing.  You'll be on your own to test the board (and likely correct errors you find), so you may be better off doing the assembly yourself anyway.

closed 2019-01-02 by mossmann

matthiasben1, 2019-01-13:
Hi!

 

Just out of curiosity, why was this issue deleted from github?

 

Also, is there a possibility to pre-order Opera Cake boards? Is there an estimated pricing?

 

Thanks,

 

-M

 

Sent: Thursday, December 27, 2018 at 5:15 PM

dominicgs, 2019-01-14:
>
> Just out of curiosity, why was this issue deleted from github?

It was closed, so it doesn't show up in the open issues list.  It's still
here: https://github.com/mossmann/hackrf/issues/568
>
> Also, is there a possibility to pre-order Opera Cake boards? Is there an
estimated pricing?

There is not currently an estimated price and no reseller is taking
pre-orders at this time.

> Sent: Thursday, December 27, 2018 at 5:15 PM
> From: "Michael Ossmann" <notifications@github.com>
> To: mossmann/hackrf <hackrf@noreply.github.com>
> Cc: matthiasben1 <matthias_ben@mail.com>, Author <
author@noreply.github.com>
>
>
> We expect to be shipping fully assembled and tested Opera Cakes by March
2019, so perhaps that will meet your need. If not. . .
>
> The "High Frequency Board" is probably your best bet, but you would need
to get stack-up details and use a microstrip impedance calculator to find
out if you need to adjust the RF trace width to achieve 50 ohm impedance.
>
> The two biggest problems you are likely to have with a prototype assembly
service for Opera Cake are characteristic impedance of the RF traces on the
PCB and lack of testing. You'll be on your own to test the board (and
likely correct errors you find), so you may be better off doing the
assembly yourself anyway.

markteo, 2019-10-14:
Where can i order an Opera Cake?

If anyone could reply this ASAP :)