greatscottgadgets/hackrf #569: hackrf suddenly not detected by USB

yuriskyuri, 2018-12-29:
### Steps to reproduce
My HackRF stopped working 2 days ago. when I plug it to USB sometimes I see RF light on, then I see this in dmesg:
[4089100.315805] usb 2-1.7: new high-speed USB device number 81 using ehci-pci
[4089100.401941] usb 2-1.7: New USB device found, idVendor=1d50, idProduct=6089
[4089100.401948] usb 2-1.7: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[4089100.401951] usb 2-1.7: Product: HackRF One
[4089100.401954] usb 2-1.7: Manufacturer: Great Scott Gadgets
[4089100.401958] usb 2-1.7: SerialNumber: 0000000000000000325866e6214c8123
[4089109.613245] usb 2-1.7: USB disconnect, device number 81

# lsusb
Bus 002 Device 081: ID 1d50:6089 OpenMoko, Inc. Great Scott Gadgets HackRF One SDR
Bus 002 Device 003: ID 0bda:0181 Realtek Semiconductor Corp.
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 003: ID 0461:0010 Primax Electronics, Ltd HP PR1101U / Primax PMX-KPR1101U Keyboard
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
You have mail in /var/spool/mail/root
#

but after few seconds device dissapears. 

other time just 3v3 rx and tx lights are on (rx and tx are very weak). I tried it on multiple computers, switched few USB cables, and it did not help, and I was using this hackrf for few weeks. any idea what could be causing this ?

# hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
No HackRF boards found.
#

yuriskyuri, 2018-12-29:
after I finished writing this, I checked device once more with external powered USB hub, same behaviour. Since it should be internal powersource issue (3v3 works but 1v8 does not), I decided to open it and check it visuallly (shop that sold me this one is closed next week). after opening it and checking all looked ok, plugged it once more, and now it works. now I am puzzled. do you have any idea which component might be failing here ? I dont consider this as normal behaviour, and I expect that I will soon need to replace some internal PSU part. but maybe you seen this before and can advice where to look.

dominicgs, 2019-01-02:
There's a possibility that this is related to USB power saving.  Could you try the recommendations here: https://github.com/mossmann/hackrf/wiki/FAQ#power-saving-and-usb-autosuspend  ?

yuriskyuri, 2019-01-03:
Hi 

I dont use this on laptop and I dont have any power saving package enabled on my desktops. when running from powered USB hub without upstream USB connection, at times when it does not work only 3v3 light light on, when board is in good mood all light expect usb lights up (and when I connect hub to PC, also USB light lights up). so this is definitelly not linux issue, something is messing power sources on board. also when board is not working at all its not detected any way on windows, in normal operation its detected somehow (and searching for drivers...)

miek, 2019-11-27:
Sorry for the long delay in getting back to you, I'm just getting caught up on the backlog of issues.

Are you still having this problem? If so, could you do a full firmware/CPLD/host tool update and see if it still persists, and update us on the latest symptoms. Thanks!

yuriskyuri, 2019-11-27:
hi 

problem still persists, I dont think this can be solved with FW upgrade, this is issue of power supply on board. from time to time, hackrf after connecting to power supply does bring up only 3v3 led, but nothing else. after few hours it suddenly starts working again. I think some power supply chip might be faulty/unstable and should be replaced.

miek, 2019-11-27:
The reason I suggest trying a firmware upgrade is that more recent firmware fixed a startup glitch that causes symptoms like this: https://greatscottgadgets.com/2018/02-28-we-fixed-the-glitch/

yuriskyuri, 2019-11-28:
interesting. I have 2nd hackrf that has this issue 100% of time (means it never starts, with same behaviour). any suggestion how to flash firmware there in that case (it does not start in any more correctly :( )

miek, 2019-11-28:
Have you tried starting it in DFU mode (hold the DFU button while plugging in the USB cable)?

jaumeqb, 2020-02-03:
Hi everyone

It happened the same to me,except mine hasn't worked again.
Did you find what failed? Did you manage to fix it?

Thanks

nbserver, 2020-04-18:
Got the same with new device. When i plug usb cable i got 3V3, PF and 1V8 leds on, nothing more. Sametime when device is connected few hours USB led is switch to ON for 2 minutes, lsusb see this device, hackrf_info too but after few minutes USB led off and nothing more. Where is problem? I tried few computers, few cables

miek, 2020-04-20:
@jaumeqb & @nbserver please open new issues with full details of the symptoms you're seeing and what you've tried. There are several different causes for USB issues, so it's confusing to debug different cases together when they may be unrelated.

I'll close this for now as we've not heard from the original poster in a while.

closed 2020-04-20 by miek