greatscottgadgets/hackrf #573: HackRF crashes when close to transmitter?

25141928, 2019-01-10:
I am using the HackRF to conduct training for a project and I am having issues recording the transmissions using multiple softwares!  Main issue:

### Steps to reproduce
1. Transmit within 1 m of the hackrf

### Expected behaviour
Hackrf records the transmission using URH software

### Actual behaviour
The HackRF records the first couple of seconds and then crashes.  Could the power of the transmitter be too much for the hackRF causing it to crash?

### Version information
Kali Linux running updated version of hackrf purchased 2018

**hackrf_info output:**

Universal Radio Hacker
Version: 2.5.4

mossmann, 2019-01-10:
How do you know it has crashed?  What are the symptoms?

Does the crash occur when the transmitter is not active?

25141928, 2019-01-10:
Thanks for replying. I should of been a bit more specific. When I transmit on the radio the software crashes when I am recording only and has to be restated. It happens on a couple of different softwares so the hack is the only single point between them. When I lower the power it doesn’t. Is this a safety feature of the hack if the power at the receive ae is too high? 

I appreciate the reply. I was hoping to be able to speak to you about the project I am working on as part of my degree. We are using the hackrf to disrupt data communications using both record and replay and packet replication. I was hoping to discuss with you your thoughts and best methods for packet replication such as within gnu radio. 

Your videos have been very useful. Thank you. 

Regards

Chris Fearn
UK

Sent using the GMX mail app

mossmann, 2019-01-10:
What is the output power of the transmitter? How close is it to the HackRF?  Is it connected by cable?  Antennas?

25141928, 2019-01-10:
It’s a couple of meters away on 15W high 5W low. Antenna only. Not cabled.

mossmann, 2019-01-10:
It sounds like you may have exceeded the maximum input power of HackRF One which can cause permanent damage:
https://github.com/mossmann/hackrf/wiki/HackRF-One#receive-power

I suggest cabling the transmitter to the HackRF One with an attenuator (50 to 80 dB) in between.

25141928, 2019-01-10:
Will do thank you for the information

closed 2019-01-10 by 25141928